OllisGit/OctoPrint-DisplayLayerProgress #91: Displayed Fan speed is 1% lower than setted in slicer

ochm, 2019-10-24:
Hi, please, can you correct displayed fan speed if  speed>0 to +1?
Problem is described here https://github.com/StefanCohen/OctoPrint-Dashboard/issues/104#issuecomment-545778417
Thank you.

OllisGit, 2019-10-24:
Hi @ochm,

I looked into the G-Code Marlin Documentation (http://marlinfw.org/docs/gcode/M106.html) and a value of 127 represent 49% and a value of 128 = 50%.

But Cura generates for a 50% Fanspeed this GCode: `M106 S127.5` and unfortunately my plugin doesn't accept float values.

I will correct this in the next version.

Thx, for reporting
Olli

OllisGit, 2019-10-24:
Okay, there was an other issue. This is fixed now in newest Version 1.14.1.

Please test and give feedback!

StefanCohen, 2019-10-25:
Confirmed working in 1.14.1

closed 2019-11-07 by OllisGit