OllisGit/OctoPrint-DisplayLayerProgress #92: Layer count are not correct

ManuGithubSteam, 2019-10-24:
Hi

Layer count is not correct.
This haunts me like since the last bug report from me. 

I personally think you should use the layer count from the lcd itself.
Just dont make if over complicated...

Anyhow, gcode is attached.
Also the print before this print, the layers where correct at the start but over the course of the print the layer count diffented from octoprint (sidebar) and the dashboard.

[TevoCube3_1h45m_DruckerRhino - Marble PLA - 0.07mm ULTRADETAIL - MK3S - Volcano 0.4.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3768971/TevoCube3_1h45m_DruckerRhino.-.Marble.PLA.-.0.07mm.ULTRADETAIL.-.MK3S.-.Volcano.0.4.zip)
[image]

OllisGit, 2019-10-24:
Hi @ManuGithubSteam ,
I uploaded your file and couldn't reproduce the error.

In your screenshot I see a lot of issues:
- Layer-Count in sidebar and in Dashboard is not correct. Should be something like 17/475
- The current height is also not correct in the dashboard

Again, you are not using in the sidebar the output of DisplayLayerProgress-Plugin. This is one reason why you see differences, BUT this difference could be only a few layers/millimeters. 

So, from my perspective one error is in the "LayerDisplay-Plugin" (Layer: **477** / 476).
And the other error is related to the Dashboard or the communication between DisplayLayerProgress-Plugin and the Dashboard-Plugin. 

More technical insides:
- DisplayLayerProgress-Plugin evaluate the Layers/Height and sends this value to the LCD, to the sidebar and also to a so called eventBus.
- Each other installed plugin could listen on this eventbus and graps the Layer/Height information (this is how the dashboard-plugin works)

Advise:
- Disable LayerDisplay-Plugin
- Enable "Show current height/ layer information in Statusbar" in Plugin-Settings
- Enable "Event debugging" in Plugin-Settings
- ReRun a print. You could use the DryRun-Plugin (https://github.com/OllisGit/OctoPrint-DryRun), so then you don't waste filament.
- After the print attach the "plugin_DisplayLayerProgress_events.log" to this issue. You can download the log from the Logging-Section.

Maybe I got you wrong...but you print two objects after each other. No server shutdown between, right!? Also in the first print the layer/height values were wrong or only in the second print?

I think we should involve @StefanCohen ....build special plugins for you with more logging informations.

StefanCohen, 2019-10-24:
Hi!

I can't reproduce this with the supplied gcode. The dashboard displays exactly what DisplayLayerProgress sends over the event bus thru the entire job.

There is however something very strange with that gcode. The height fluctuates almost 0.5mm during each layer as you can see here: https://imgur.com/a/LNi2u24

//Stefan

OllisGit, 2019-10-25:
There is an other strange thing in the gcode:
[image]

I thing the "BEFORE_LAYER" M117 message is created by your slicer and the following M117 is created by my plugin. Because the layer-expression detects ";LAYER:474"

ManuGithubSteam, 2019-10-25:
Hi,

I will be away at the weekend but will give you the logs next week when i
get back from my trip.

I use she standard prusaslicer 2.1 I'm not sure if I did some gcode message
alterations to make it work but I think I removed them again because before
layer seemd to have worked ok.

Anyway I will check.

How can the layer bounce like that? The print comes out just fine...

If you need a Teamviewer session to debug this just let me know.

Have a nice weekend

Manuel

StefanCohen, 2019-10-25:
@ManuGithubSteam Can you please re-slice and attach the unprocessed gcode as it comes straight out of the slicer (before it has been uploaded to OctoPrint?) I think that may help pinpointing the problem.

ManuGithubSteam, 2019-10-27:
[Reclice.tar.gz](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3775806/Reclice.tar.gz)
There you go. (Old Gcode from above)

New piece:
[PaperHolder v5_10m_DruckerRhino - Marble PLA - 0.20mm QUALITY - MK3S - Volcano 0.4.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3775814/PaperHolder.v5_10m_DruckerRhino.-.Marble.PLA.-.0.20mm.QUALITY.-.MK3S.-.Volcano.0.4.zip)

Advise:

Disable LayerDisplay-Plugin
--> Do not use this plugin or could not find it.
Enable "Show current height/ layer information in Statusbar" in Plugin-Settings
-> done
Enable "Event debugging" in Plugin-Settings
-> done
ReRun a print. You could use the DryRun-Plugin (https://github.com/OllisGit/OctoPrint-DryRun), so then you don't waste filament.
-> printing at the moment
After the print attach the "plugin_DisplayLayerProgress_events.log" to this issue. You can download the log from the Logging-Section.

StefanCohen, 2019-10-27:
I loaded your gcode in PrusaSlicer and noticed that you have Lift Z set to 0.2 mm (in Printer Settings->Extruder1). That explains the fluctuating height as it is constantly lifting the head whenever there is a retraction. Here's an example from Layer 2:

```
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.28
;LAYER:1

G1 E-0.16000 F4500.00000
G1 F6240
G1 X116.146 Y106.254 E-1.36800
G1 E-0.07200 F4500.00000
G1 Z0.400 F7800.000
;AFTER_LAYER_CHANGE
;0.28
G1 X109.467 Y109.840
G1 Z0.280
G1 E1.60000 F4500.00000
G1 F2963 
```

OllisGit, 2019-10-27:
I looked into the eventlog and into the gcode and found strange things that I can not explain.
- The layer message command is send twice for each layer (see below layer 5)
- The layer 5 message is send before layer 4 message is send 
```
2019-10-27 13:17:16,592 SENDING-HOOK: M117 INDICATOR-Layer5 [1672406128::comm.sending_thread]
2019-10-27 13:17:16,596 UPDATE DISPLAY: layerChanged [1672406128::comm.sending_thread]
2019-10-27 13:17:16,636 SENDING-HOOK: G92 E0.0 [1672406128::comm.sending_thread]
2019-10-27 13:17:16,642 SENDING-HOOK: M117 INDICATOR-Layer4 [1672406128::comm.sending_thread]
...
2019-10-27 13:18:59,027 SENDING-HOOK: M117 INDICATOR-Layer5 [1672406128::comm.sending_thread]
```

Please so kind and make screenshots of the settings-tab: Display and Layer

...at the moment I have no idea, why this happens....maybe I need to create a special version of my plugin to log more details.

ManuGithubSteam, 2019-10-27:
Here you go:

[image]
[image]
[image]

StefanCohen, 2019-10-27:
It seems like you have added`;LAYER:[layer_num]` to your before layer change G-code. That is probably what throws DisplayLayerProgress off as that is a layer indicator for another Slicer (IdeaMaker) if I'm not mistaken. Comment it out, re-slice and try again.

ManuGithubSteam, 2019-10-27:
Ok will try that. have a big print running atm. You sould more clearly state, what needs to be added to the section to get it to work correctly :-)

OllisGit, 2019-10-27:
@StefanCohen good catch!
```
;BEFORE_LAYER_CHANGE
M117 INDICATOR-Layer1
G92 E0.0
;0.2
;LAYER:0
M117 INDICATOR-Layer0
```
In the sliced gcode there are two indicator commands: ```;BEFORE_LAYER_CHANGE``` and then ```;LAYER:0```
You have two option:
- remove one of the expression in my plugin
- remove one entry in your slicer

...but... don't think that solves the initial problem...maybe!

StefanCohen, 2019-10-27:
The default "Before Layer Change gcode" in Prusaslicer is:

```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
```

You seem to have manually added the line `;LAYER:[layer_num]` to that block. As Ollis said, either:

1. Remove that line or remove the matching regexp line in the Layer settings of DisplayLayerProgress. Then restart octoprint and re-upload the gcode.
2. Comment out or remove `;LAYER:[layer_num]` in Prusaslicer. Then re-slice and re-upload the gcode.

OllisGit, 2019-11-22:
No feedback since a while -> closed

closed 2019-11-22 by OllisGit