OllisGit/OctoPrint-DisplayLayerProgress #94: Layer Progress not shown after Filament Change ( MARLIN routine )

DennisHICTOP, 2019-10-31:
Printer=HICTOP, Octoprint=1.3.12, Octopi=0.16.0, Firmware MARLIN 06.13.16 origin unknown
Issue with Dashboard plug-in.
On a fresh booted system the bottom line of the Printer display correctly shows the Layer info from Dashboard, and is also on the bttom left of PC Octoprint page.
All works as it should unless on the Printer Board(RAMPS)/display, I select from the TUNE menu, the last item "Change filament".
On selecting this, (after some seconds, I assume until buffer of instructions is exhausted )_ the head moves to home position, filament is reversed out, and it sits there. On manually feeding new fillament into the extruder, and pressing the physical button on the printers display, the head resumes printing.
BUT the bottom line of the display continues to show    Change filament for this and subsequent prints.

The Octoprint page (on bottom left) of PC continues to show the correct situation for layers,

Starting a new print does not remove the   change filament    message on the bottom line of the printers display, and the PC page continues to behave normally, showing the correct progress.

[image]
[image]

However, on the PC, selecting Reboot System ( while printer is idle ! ) from the power options of Octoprint, restores the bottom line to the dashboard info. I have not tried the other option yet - Restart Octoprint.

Attached are:-
IMG_0197 = Printer display prior to filament change
IMG_0196 = Printer display after filament change
Slicer_Facis.txt  the gcode from Simplyfo3D v4.0.0
Octopring_Facia.txt = the gcode downloaded from Octoprint ( running on Raspberry Pi )

[Slicer_Facia.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3795073/Slicer_Facia.txt)
[Octoprint_Facia.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3795074/Octoprint_Facia.txt)

191031_DennisHICTOP_OllisGit

OllisGit, 2019-11-02:
Hi @DennisHICTOP ,

from the gcode perspective everything looks good. So, my first idea is that the marlin firmware makes "some tricks".

Is this error "repeatable"?

If you print the same file with deactivated plugin and manuel filament-change, is the display showing the last M117 message: ```M117 Job Done - Enjoy !```?

I looked for similar bugs for your firmware, but I couldn't found your version (06.13.16) in the "vendors-git".
Is it possible to upgrade your firmware, is this an option for you?

Maybe you can also raise an issue in the marlins issue-tracker or ask in a 3dprinter forum, about sending M117-messages after manuel filament change.

DennisHICTOP, 2019-11-11:
OllisGit,
The issue seems to be with MARLIN.
It happens with Octopi disconnected, and printing from SD card.
the Marlin, I think is 1.1.0 or 1.1.6
As I use this mainly to change colour at a specific layer, I stand over the printer watching the bottom line of the display, and at the appropriate time select Change Filament from the Marlin UI.
So, to make my life easier, and circumvent the issue, I have written a VB app to allow me to input the layer(s) (x's) I want to have a change, I then parse the gcode file for  ; layer "x" , Z = 
If x is in the input list, then I insert gcode to raise extruder, move to near home, eject filament, and finally M0 ( which changes the Pause button in Octoprint to Continue)
This modified file is then sent to Octoprint
Pressing the Continue button in Octoprint  resumes running the gcode file from the M0 instruction.
[image]
If 8 changes are not enough, I run it again on the same file.

I am happy now, and don't need to start the risky business of Flashing the controller, as its far too easy to brick it, if I get it wrong.
Thanks for the help, and the thought/effort you have given. Always nice to know there is always someone who know more.

OllisGit, 2019-11-11:
Hi @DennisHICTOP,
good to know that you found a solution for you.
I asked the same question in reprap-forum, but without success: https://reprap.org/forum/read.php?415,861740,861742#msg-861742

Modifiing the gcode could not be done by your  slicer (like pre-processing)?
Or was it easier to code a VB-Script?

OllisGit, 2019-11-22:
Nothing to do here -> closed!

closed 2019-11-22 by OllisGit