OllisGit/OctoPrint-DisplayLayerProgress #95: Page crashes on iPad

falvesjr, 2019-11-08:
After much testing today, I've come to the conclusion that the latest version of DLP (1.14.2) is causing the OctoPrint pages to crash on iPad browsers after about 10-15 seconds. As the issue only happened when OctoPrint was actually printing, I suspected DLP might be the culprit as I remembered that there was an update a couple of days ago. I downgraded DLP to 1.14.1 and the problem went away.

Could you please take a look and confirm that DLP is causing problems on iOS?

Thanks,
Fernando

falvesjr, 2019-11-08:
[image]
[image]

OllisGit, 2019-11-08:
Hi @falvesjr ,

I don't have an ios device, but I will do my best to analyse the issue (maybe with your help).

BR
Olli

falvesjr, 2019-11-08:
Hi Olli,

I'll be happy to help. Let me know what I can do.

I would start by looking at what changed in your last update as 1.14.1 works fine, but 1.14.2 does not.

Cheers,
Fernando

OllisGit, 2019-11-08:
Hi @falvesjr ,
the only difference between the two versions is detecting movement mode (real, abs) on the server-side. So, there was no UI-Change.

[image]

Do you know the timestamp of the crash, if yes please provide the octoprint log file (Settings / Logging / Download)

If not, please do the following:
- Install my "DryRun" plugin, so that you don't waste filament during testing
- Activate debugging in DLP-Settings page
- Use your ipad and start printing
- After "crashing", provide the octoprint-log and the dlp-event log file

Is the octoprint webpage available from an other system (laptop, pc, mobile, tv, ...), after the crash?

falvesjr, 2019-11-08:
Olli,

I have a print running that won't be finished until tomorrow, but I have some logs I saved yesterday and I'll attach them here.

Perhaps the word "crash" is too strong... It's more like an error which prevents browsers on iOS from displaying the page, OctoPrint itself is fine and is not affected at all. The pages also work fine on my Mac, but I tend to use my iPad a lot to check on prints and would like to keep doing that.

Also, it doesn't matter if I start printing from my Mac or from the iPad, the problem appears when an active print is running. Maybe the problem is also related to the Dashboard plugin? I don't think so because I haven't changed the Dashboard plugin, but going back to DLP 1.14.1 stopped it happening.

Anyway, let me know if you can see anything on these logs, if not, I'll try to do what you asked after my print ends tomorrow.

[octoprint.2019-11-04.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3825028/octoprint.2019-11-04.log)
[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3825029/octoprint.log)

OllisGit, 2019-11-08:
Do you have a timestamp for me? I see a lot of warnings that the server could not send informations to the UI (tablet). This seems to be related to the ForceLogin-Plugin, maybe this could also be good test.
- Deactivate the Force-Login Plugin and retry.

Btw: I tested with the current Dashboard-Plugin on a simulator ipad without any issue:
[image]

falvesjr, 2019-11-08:
I can't try disabling the forcelogin plugin right now as I'm in the middle of a print until tomorrow afternoon, but here's attached a more recent log and at 2019-11-08 11:33:46,909 you can see that I logged in with my iPad and all is normal.

It does sound like the forcedlogin plugin might be causing the problem (I don't even know what it does), but why does it only cause the problem with DLP 1.14.2 and not with 1.14.1?

[octoprint.2019-11-08.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3825630/octoprint.2019-11-08.log)

falvesjr, 2019-11-08:
Also, a simulator might not be the same as the device. I don't know if it uses the same version of WebKit and what version of the OS is it running? My iPadOS is 13.2.2 on an iPad Pro 9.7-inch device. And Safari on iPadOS is a little more tolerant than Chrome to this bug and may survive a little longer before displaying the "repeated problem" alert, maybe 30-60 seconds. Chrome only tales about 10-15 seconds.

OllisGit, 2019-11-23:
Could you open the developer tools in safari or chrome? If yes, check the box preserve log and wait for the crash. Maybe you can see some error message in the console.

OllisGit, 2020-04-18:
No feedback...and I could not reproduce the issue, because I don't have an ipad (yet)

closed 2020-04-18 by OllisGit