OllisGit/OctoPrint-DisplayLayerProgress #96: Support of more progress values like:

OllisGit, 2019-11-21:
start/endtime
duration
time to finish

OllisGit, 2020-01-19:
```[estimated_end_time], [printtime_left]``` is now in version 1.17.0

Wob76, 2020-01-21:
Hi, Thanks for the extra values, I am really liking the new tab feature. It would appear that [estimated_end_time] is showing as GMT rather than the local timezone, I have it running on 2 printers with the same result, I am at +11 and they are both showing off by the same value.

OllisGit, 2020-01-21:
Hi @Wob76,
thx for the feedback.

I created a new issue for that feature: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/108

OllisGit, 2020-01-30:
Hi @Wob76 ,
new Version 1.17.1 is out. The timezone and 24h is fixed (hopefully, couldn't test it. I just implemented like in the python doc).

Please test and give feedback!

Thx, in advance
Olli

Wob76, 2020-01-30:
Thanks @OllisGit I'll update and test as soon as my current print is done.

StefanCohen, 2020-02-04:
So you mean I don't have to calculate the ETA myself anymore? Perfect! 👍

Wob76, 2020-02-04:
@OllisGit As reported on #108 the ETA timezone is working as expected now after the update and ensuring I had the TZ of the pi set correctly.

OllisGit, 2020-02-05:
Hi @StefanCohen ,
yes ```'estimatedEndTime':'20:24'``` is available since V1.17.0.

And in the next release (1.18..0) ```estimatedChangedFilamentTime``` will be also available ;-)

closed 2020-02-05 by OllisGit