OllisGit/OctoPrint-DisplayLayerProgress #98: When used with OctoPrint-UltimakerFormatPackage, layers are not calculated

RomRider, 2019-12-03:
When I send the job to octoprint using Cura with UltimakerFormatPackage installed and the Octoprint intergration in Cura, Cura sends the file in UFP which includes a thumbnail of the object. The gcode is then extracted by UltimakerFormatPackage.
In this case, DisplayLayerProgress doesn't parse/modify the gcode and thus can't display layer progress.

Is that something which could be implemented?

redpizza69, 2019-12-22:
That's what the problem is! I've been spending hours trying to work out why DisplayLayerProgress isn't working!

+1 for working with the UltimakerFormatPackage from Cura.

OllisGit, 2020-01-02:
Hi @RomRider , @redpizza69 
this issue is solved in the newest version 1.16.0.

Please test and give feedback!

OllisGit, 2020-01-19:
Hi @RomRider , @redpizza69
the new version 1.17.0 is also able to handle ufp-format.

Please test and give feedback!

Thx, in advance
Olli

OllisGit, 2020-01-30:
New version, new luck!

closed 2020-01-30 by OllisGit