OllisGit/OctoPrint-DisplayLayerProgress #99: File selection relies on long deprecated (and now removed) field 'file'

foosel, 2019-12-06:
Hey there! I just got a report by @loskexos of an error triggered in your plugin on 1.4.0rc2 here: https://github.com/foosel/OctoPrint/issues/3360#issuecomment-562610634

```
2019-12-06 09:49:33,196 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 44127}
2019-12-06 09:51:22,962 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: BagClip_85_0.1mm_PLA_MK3_1h12m.gcode, owner: octoadmin, user: octoadmin
2019-12-06 09:51:22,974 - octoprint.plugins.DisplayLayerProgress - INFO - File selected. Determining number of layers.
2019-12-06 09:51:23,019 - octoprint.plugin - ERROR - Error while calling plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 344, in on_event
    with open(selectedFile, "r") as f:
IOError: [Errno 2] No such file or directory: ''
2019-12-06 09:51:23,037 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2019-12-06 09:51:23,139 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: BagClip_85_0.1mm_PLA_MK3_1h12m.gcode, owner: octoadmin, user: octoadmin
2019-12-06 09:51:23,148 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-12-06 09:51:23,184 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2019-12-06 09:51:23,195 - octoprint.plugin - ERROR - Error while calling plugin cancelobject
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 250, in on_event
    with open(selectedFile, "r") as f:
IOError: [Errno 2] No such file or directory: ''
```

Taking a look at your code you are retrieving `selectedField` from the event payload of `FileSelected` and more specifically the deprecated field `file`, which was marked as deprecated in 1.3.0 and removed in 1.4.0:

[image]

([Docs in question](http://docs.octoprint.org/en/master/events/index.html#file-handling), reason for removal is that we really should not be leaking full disk paths all over the place with events).

You should switch that to use `path` and `origin` instead and resolve those to the path on disk using `self._file_manager.path_on_disk`, otherwise your plugin will run into issues with 1.4.0.

OllisGit, 2019-12-21:
Hi @foosel , @loskexos ,
the bug is fixed in newest Version 1.15.0

Please test and give feedback
C ya,
Olli

loskexos, 2019-12-21:
@OllisGit : Error message vanished after going to displaylayerprogress:1.14.2. Thank you!
(1.14.2 was showing as available update, not 1.15.0)

OllisGit, 2019-12-22:
Hi @loskexos 
if you run DLP 1.14.2 in a OP 1.4.0RC it could not work! Because this version use old/not provided methods from octoprint 1.3.x

Regarding not showing the newest version: It could be that your local plugin-cache prevents a new lookup to the repositories.
Go to the "Software Update"-section and click the "wrench-button", look into the TTL-Value.

Also, click on "Advanced options" (tiny text below) and then press "Force update". This will refresh the cache. An auto update is not executed, but an "Update available"-popup will appear.

loskexos, 2019-12-22:
@OllisGit ... just took the last displayed Version Nr from logs. 
Upon checking - actual version in pluginmanager is 1.15.0. Sorry about that.

closed 2019-12-23 by OllisGit