OllisGit/OctoPrint-DisplayLayerProgress #100: Error initializing plugin

ChronoStriker1, 2019-12-20:
I am getting this error when ever I attempt to enable the plugin on octoprint 1.3.12
I should note that I had updated to 1.4.0rc3 but downgraded back to 1.3.12 once I realized this plugin is not yet compatible with it.

2019-12-20 11:56:12,053 - octoprint.plugin.core - ERROR - Exception while initializing plugin DisplayLayerProgress, disabling it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1358, in initialize_implementation
    implementation.initialize()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 202, in initialize
    self._evaluatePrinterMessagePattern()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 631, in _evaluatePrinterMessagePattern
    if PROGRESS_KEYWORD_EXPRESSION in printerMessagePattern:
TypeError: argument of type 'NoneType' is not iterable

OllisGit, 2019-12-21:
Hi @ChronoStriker1 
it looks like that the plugin-settings were corrupt.

Please look into the ```config.yaml``` file for the section ```plugins:  DisplayLayerProgress:```
Maybe you can identifiy something, if not you need to reinstall the plugin.
In that case you should uninstall the plugin via UI, check config.yaml for plugin-fragments, restart the server and then reinstall the plugin.

Next version is in progress to make it run with 1.4.x and python3

BR
Olli

OllisGit, 2019-12-21:
Hi @ChronoStriker1 
the newest Version 1.15.0 is now running with OctoPrint 1.4.0RC3

Please test and give feedback
C ya,
Olli

ChronoStriker1, 2019-12-22:
I ended up setting up a new SD card for other reasons but I did upgrade and can confirm it is working on 1.4.0RC3

closed 2019-12-22 by ChronoStriker1