OllisGit/OctoPrint-DisplayLayerProgress #101: Large upload, 56Mb, fails

Taomyn, 2019-12-24:
When I try to upload some gcode, a 56Mb file, via the browser it gets passed the upload stage and during the saving part it fails

[image]

When the server returns as I assume it must be restarting, the file is present. If I disable the plug-in and restart the upload works without error.

Using OctoPrint 1.3.12 (with OctiPi image on Raspberry Pi-3B), all updated, and plug-in is 1.15.0

OllisGit, 2020-01-02:
Hi @Taomyn ,
this should be fixed in the newest version 1.16.0, because the plugin doesn't analyse/modify the g-code during upload. Instead it use now the file-selection event.

Please test and give feedback!

Taomyn, 2020-01-03:
Thank you very much. I updated and was able to upload a 44Mb file which I know failed the other day - it was so much faster as well.

OllisGit, 2020-01-19:
Hi @Taomyn,
in the new release v1.17.0 I switched back to the upload approach, because of other issues.
BUT I could improve the analyse performance up to 50%, so uploading of big files >50MB should now be possible.

Please test and give feedback.

Thx, in advance
Olli

Taomyn, 2020-01-20:
@OllisGit thanks, I updated remotely without issue but I don't have any large gcode to upload at the moment. Will generate some tonight and test it out.

Taomyn, 2020-01-21:
Sorry for the delay. I just tried to upload a 101Mb gcode file and it timed out with the same error.

OllisGit, 2020-01-25:
Hi @Taomyn,
I created a forum-entry to ask the community about timeout issue.
https://community.octoprint.org/t/timeout-during-local-upload-of-gcode/15487

To analyse the error I have some additional questions:
- Do you have some kind of ReverseProxy (e.g. Apache, nginx) in front of OctoPrint-Server? Or do you call octoprint directly from the browser (e.g. with IP-Adress)?
- Which browser do you use? Did you try different browsers?
- hmm..connected via WiFi or Cable?
- Do you have the option to switch to a more powerfull raspberry or desktop-pc?

Taomyn, 2020-01-26:
> Hi @Taomyn,
> To analyse the error I have some additional questions:
> 
> * Do you have some kind of ReverseProxy (e.g. Apache, nginx) in front of OctoPrint-Server? Or do you call octoprint directly from the browser (e.g. with IP-Adress)?

You've hit the nail on the head with this question, I now use HAProxy through my firewall to access it and when I tested the update it failed. It did however fail when I first reported the issue as the proxy was not in play. I used the direct IP and it was fine.

Needed to changed the HAProxy "server timeout" to 60s and now it's fine a 101MB file was ok. Sorry about that but as it was exactly the same problem as before i.e. uploaded fine but the saving would time out, it didn't click that it was the proxy this time.

> * Which browser do you use? Did you try different browsers?

It didn't matter which browser

> * hmm..connected via WiFi or Cable?

WiFi
> * Do you have the option to switch to a more powerfull raspberry or desktop-pc?

Nope, it's the only spare I have

OllisGit, 2020-02-05:
Hi @Taomyn,
fyi: I just want you to inform that I am still working on this issue. My current solutions looks like this:
- There will be a size-threshold(default 50MB)
- If the limit is reached, the layer analyse is not started directly, instead it is started in the background with in an other "process"

The current challenge I have is to block the "start-print-function" during the background analyse and release the lock after the analyse is done.

BR
Olli

OllisGit, 2020-04-08:
Hi @Taomyn,
in the newest release you can disable the modification during upload. But then you need to add the Layer-Indicator-command by your own in your slicer.
See https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.19.0

Please test and give me feedback.

Thx, in advance
Olli

closed 2020-08-15 by stale[bot]