OllisGit/OctoPrint-DisplayLayerProgress #102: No Layer / Total with Prusa-Slicer, "LayerDisplay" works

chatelao, 2019-12-26:
Problem:
- The Layer stats show -/- with DisplayLaye.rProgress

The alternative plugin "LayerDisplay" (and the built-in gCode Overview) work both fine:
- https://plugins.octoprint.org/plugins/layerdisplay/

The gCode used is here:
- https://gist.github.com/chatelao/563d919a3bbb4ee29c6314c0b0bcf2fb

OllisGit, 2019-12-27:
Hi @chatelao ,

what layer-indicator and what layer-expression (plugin-settings) do you use? Could find it in your g-code.
Please look into the readme from the start-page.

chatelao, 2019-12-27:
Dear @OllisGit I just installed it "as-it-is". But will have a look at the Readme again.

OllisGit, 2019-12-29:
@chatelao 
Sorry, was a typo: ```Couldn't find...```
You need to add this information in your slicer, like this:
[image]
After that, everything should work out of the box.

chatelao, 2019-12-30:
An API from the Code, just exposing the same numbre would be great. Let's as Gina, I will be back, if there is a simple answer.

OllisGit, 2019-12-30:
Hi @chatelao!
I didn't get it! What do you mean?

There is an REST-API to get all values, read the [developer-section](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress#developer---section) from the readme.md!

chatelao, 2019-12-30:
Hi @OllisGit , I just wondered if you could get the layer information from the OctoPrint core, instead of analyzing yourself. - This would remove (move to core) the dependency to the slicer.

OllisGit, 2019-12-30:
Hi @chatelao unfortunately this is not possible, because octoprint doesn't provide such an api to get the necessary informations. 

Yes, there is the clientside javascript based GCode-Viewer, but there is also no hook to catch the layer information.
One option could be to grap the layerInfo from the bottom gcodeviewer-legend and parse that text, but in my option this is a hack and also this box show not always the correct layer information.
Only the slicers knows the exact layer-infos!

closed 2020-01-02 by OllisGit