OllisGit/OctoPrint-DisplayLayerProgress #104: Plugin not working after update today.

samd12, 2020-01-02:
I get layer count but no progress reporting back

iherbak, 2020-01-03:
Percentage and total layer are there but current layer is only a dash symbol

OllisGit, 2020-01-03:
Hi @samd12 , @iherbak ,

sorry to say, but I can't reproduced the mentioned behaviour.

I installed a fresh version 1.16.0 on my printer, printed already uploaded files and also I uploaded a new file. Each time the navbar, desktop and printer display shows me the correct output.

To analyse the issue I need more informations:
- do you upgrade to the newest version or did you install a fresh copy of the newest version ?
- which octoprint version do you use ?
- please download your gcode directly from octoprint and check for the layer indicators. Something like this:
```
;LAYER:0
M117 INDICATOR-Layer0
```
- make a screenshot of the Display-Tab so that I can verify that you use the correct keys for the values or do a "Reset Settings" and try again
- look into the octoprint logfile (accessible via settings menu) for errors, if your unsure you can attached a zip-file to this issue
- activate the debugging log via the plugin-settings on the right tab. After that, start a print and after the error occurs you can stop the print (optional), download the ```plugin_DisplayLayerProgress_events.log``` and attach it to this issue.

For the test prints you can use my [Dry-Run Plugin](https://github.com/OllisGit/OctoPrint-DryRun), so you don't waste energy and filament.

Thx,
Olli

iherbak, 2020-01-03:
Hi, here aer the details of my setup.

[plugin.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4020503/plugin.zip)

OllisGit, 2020-01-04:
Hi @iherbak ,

thanks for the files. The problem is that the layer-indicators were not generated in your g-code. 
The provided g-code doesn't include such M117 commands:
```
M117 Printing...
; process Process1
; layer 1, Z = 0.200
T0
G92 E0.0000
```
After selecting the file in octoprint the plugin replace the layer-comments (```; layer 1, Z = 0.200```)with layer indicators M117-commands (```M117 INDICATOR-Layer1```), because during printing octoprint filters all comments.
So, the g-code file downloaded from octoprint should look like this:
```
M117 Printing...
; process Process1
; layer 1, Z = 0.200
M117 INDICATOR-Layer1
T0
G92 E0.0000
```
Please check your layer plugin settings. It should include:
```
1		[; layer ([0-9]+),.*]		Simplify3D
```

iherbak, 2020-01-04:
Hi,

As you can see the setting is there, but it is not adding M117. Reinstalled
the plugin but same thing happened. First time i upgraded from the previous
version when it started the issue.

[image: image.png]

Thanks

OllisGit, 2020-01-04:
Hi @iherbak ,
I can't see the image, but please tell me: Did you modify the expressions or do you use the default-values?
Good to know that you upgrade, please do me a favour: 
- open the ```config.yaml``` file on your octoprint-server (e.g. raspberry). See Folder-section if you don't know where to find such file. Octoprint settings and look into the root folder (../uploads).

- look for this section:
```yaml
plugins:
    DisplayLayerProgress:
    ...    
```
- copy/paste everything, so that I can use it in my environment

If you can't find the section, you use the default plugin-settings.

iherbak, 2020-01-04:
Hi,

Here are the requested details. I'm using the defaults
[plugin.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4021799/plugin.zip)

OllisGit, 2020-01-04:
Hi @iherbak !
hmmm...strange...I can't find any error in my environments.

In your first zip-file the ```plugin_DisplayLayerProgress_events.log``` is empty. 
So, I decided to create a new custom version with improved default logging.

Please install the current version from the master branch:

https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/archive/master.zip

After installation you should see the version number ```1.16.0ISSUE104```

Please select the file again and then send me the octoprint.log. Hopefully I can see more information.

Thx, in advance!

iherbak, 2020-01-04:
Hi,

Installed the package, verified the version. Ran a print. The log shows the replacements, also the file contains now the m117 but both the printer lcd and the octopi showing what i sent before. The actual layer is only a dash sign
[Downloads.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4022190/Downloads.zip)

OllisGit, 2020-01-04:
Hi @iherbak,
thx for the fast feedback, the log is really helpfull.

How do you start a print job?
1. select the file, wait for processing
2. press print-button

or do you use the small button "Load and print" ?

In the log-file I can see that selecting the file and starting the print is very close together (few milliseconds). 

So, my current assumption is, that the print job started with the 'not modified' gcode. Reading into memory for print is done, but the layer processing is not done.

```
2020-01-04 16:55:51,971 - octoprint.printer.standard.job - INFO - Print job selected 
2020-01-04 16:55:51,980 - octoprint.plugins.DisplayLayerProgress - INFO - File selected. Determining number of layers.
2020-01-04 16:55:52,006 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
...
2020-01-04 16:55:52,034 - octoprint.plugins.DisplayLayerProgress - INFO - Count '1'
2020-01-04 16:55:52,035 - octoprint.printer.standard.job - INFO - Print job started
2020-01-04 16:55:52,088 - octoprint.plugins.DisplayLayerProgress - INFO - Count '2'
2020-01-04 16:55:52,115 - octoprint.plugins.DisplayLayerProgress - INFO - Count '3'
...
2020-01-04 16:55:53,360 - octoprint.plugins.DisplayLayerProgress - INFO - File select-event processing done!'
```

Please try to select the file, wait a few seconds and then start the print.

iherbak, 2020-01-04:
Hi, I usually (99.9%) use the small print icon next to the file. I tried
the click first and print. Now it is displaying correctly as it should.

Seems so it is connected to the load and print.

OllisGit <notifications@github.com> (időpont: 2020. jan. 4., Szo, 18:29)
ezt írta:

samd12, 2020-01-04:
Hi OllisGit.  Sorry was tied up the past day or so.  I did a fresh install of octopi, dashboard and the plugin today.  All downloaded fresh on a new sd card.  I am still having the same issue.  I am just getting started with this stuff so I may be a bit slow to get you info as I need to learn how to do it as I go.  IT backround so there is hope ;) Can you describe exactly what you need and if you can any help on where to learn to get the data would be great.    Very much appreciated.  btw is there an easy way for me to install the previous version to see if the issue goes away??  Thanks again

samd12, 2020-01-05:
OK 

[Octo_disp_lay_upload.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4022885/Octo_disp_lay_upload.zip)
I installed a fresh load of octoprint.  Did all updates to the PI also.  So I am sitting completely up to date as of today.  My installation has a Pi3+ with a 7" display running touch.  As of right now I am not getting the layer information correctly.  I get the layer total but no status of what layer I am on.  The layer % is reporting correctly in the status bar as progress goes.  What is odd is that on the touch screen I am getting a message saying "Cant get stats from DisplayLayerProgress"  Is it installed.  It is installed.  I have uninstalled it and reinstalled it with no change.  
I have included the logs that you requested as well as some photos of settings.  It was a 10 minute print that I needed anyway.  Took me a bit to figure out how to get the logs off of the pi.  
Thanks again, let me know if you need anything else

AntinOz, 2020-01-05:
I have what appears to be the some issue. Let me know if you need further diagnostics.

OllisGit, 2020-01-05:
Hi @AntinOz ,
only one question, because I think the issue is identified: 

How do you start a print job?
1. select file and then press print-button
2. press small button "Load and print" next to the file
3. or via api-call (e.g. from your slicer)

samd12, 2020-01-05:
I usually start by hitting print in Cura (latest version)

OllisGit, 2020-01-05:
Hi @samd12,
you already mentioned that you that you start the print via Cura.  Thx, for that info!

This is currently the issue with Version 1.16.0. At the moment you have two options:

1. Don't use the auto start feature in Cura (remove autostart checkbox). Instead: Just transfer the file with Cura to OctoPrint. In OctoPrint select the file, wait a few seconds and press the print-button in OctoPrint 
2. Downgrade to Version 1.15.0. Just enter the following URL in the Plugin-Manager from OctoPrint. The current version will be overwritten with 1.15.0, settings will not be removed
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.15.0/master.zip

OllisGit, 2020-01-05:
Hi @maz323f @ngroulx @arunsenau @samd12,
the problem is identified and I am working on a solution.

The issue is related to the "layer indicator-identifier process". In 1.16.0 I changed that from fileupload to fileselection. My assumption was that the most users pick a file and after that press the print--button....but this was wrong. Starting a print job is done via small button "Load and print" or via API-Call like in Cura.

Hopefully I can find a solution for that....in worst case I need to switch back to 1.15 approach...but that UltimakerFormatPlugin will not be working....will see!

arunsenau, 2020-01-05:
Great news that makes sense!

Perhaps a way to resolve this would be to use the 1.15 approach by default and allow a checkbox for the alternate behaviour for the UltimakerFormat users.

ngroulx, 2020-01-06:
Thanks for your hard work on this @OllisGit

AntinOz, 2020-01-06:
> Hi @AntinOz ,
> only one question, because I think the issue is identified:
> 
> How do you start a print job?
> 
> 1. select file and then press print-button
> 2. press small button "Load and print" next to the file
> 3. or via api-call (e.g. from your slicer)

3. Direct from Cura

OllisGit, 2020-01-19:
Hi @samd12, @iherbak, @AntinOz, @arunsenau, @ngroulx    

all mentioned problems should be solved in version 1.17.0. 
See release news for details: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.17.0

Please test and give me a feedback. 
If you find a bug, please create a new issue. Because in this issue a lot of different things were raised.

Thx, in advance
Olli

iherbak, 2020-01-20:
Hi,

Updated and tried withmy usual routine. Upload and press small print icon.
Works as expected.
Thanks a lot
István

OllisGit <notifications@github.com> (időpont: 2020. jan. 19., V, 21:23) ezt
írta:

samd12, 2020-01-22:
Sorry for the delay, got to print today and it is working perfectly!  Thank you so much for all you do!!

AntinOz, 2020-01-22:
Printing right now, working as expected. Really appreciate this.

closed 2020-01-28 by OllisGit

firozdang, 2021-01-31:
I have the same issue whats the fix?