OllisGit/OctoPrint-DisplayLayerProgress #105: Predicted end time

chatelao, 2020-01-03:
I like the predicted end-time in the telegram plug-in. It would be nice to display this directly on the printer.

Ideally there would be an [estimated_end_time] tag, formated in the 24 hours layout ;-).

```15% 13.2/88.1mm ~21:03```

OllisGit, 2020-01-03:
Hi @chatelao ,
thanks for this feature request. It is already on my list, see #96.

Also, I want to take a look into this plugin https://plugins.octoprint.org/plugins/PrintTimeGenius/
maybe I can include this output into the printer display as well.

chatelao, 2020-01-03:
Thank you a lot!

OllisGit, 2020-01-19:
Hi @chatelao,

the new version 1.17.0 include the new feature. Just put ```[estimated_end_time]``` in your message pattern.

Please test and give feedback.

Thx, in advance
Olli

TobbeJ, 2020-01-21:
how do you modify the date format to make more sense?
currently i use octoprint from docker and i ended up with UTC timestamp and 12h clock.
is the plugin relying on the systems own time formatting and timezone? if so perhaps the tz is wrong in the container

OllisGit, 2020-01-21:
hi @TobbeJ,
a couple of minutes ago I just create a new issue related to the time format: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/108

The current implementation use the system timezone.

Is it possible for you to change the timezone in the docker-container or do you think the plugin should allow defining a timezone for the format?

OllisGit, 2020-01-28:
Hi @TobbeJ ,
from my side the recalculation of the end-time is done each time "something" is changed. E.g. progress, height, layer, feedrate, etc.
But I am grabbing the "printTimeLeft-Value" from Octoprint and add the current time to it. I don't know when this value is recalculated.

The 12h/24h issue should be solved in the next release (could not tested, just implemented as described in python-doc)

TobbeJ, 2020-01-29:
i think i was wrong about the time not updating properly.
my first test was just a short print and there probably was not enough time to update it.

i have printed a couple of much longer prints and it works fine, time gets updated now and then.
so only problem left is that the clock is in 12h instead of 24h.

OllisGit, 2020-01-30:
Hi @chatelao, @TobbeJ ,
new Version 1.17.1 is out. The timezone and 24h is fixed (hopefully, couldn't test it. I just implemented like in the python doc).

Please test and give feedback!

Thx, in advance
Olli

chatelao, 2020-01-31:
Great Feature, thank you

TobbeJ, 2020-02-03:
Appears to work as expected now.

closed 2020-02-04 by OllisGit