OllisGit/OctoPrint-DisplayLayerProgress #106: v 1.16.0 problem

maz323f, 2020-01-03:
After updating the plugin today. it only shows % of the print job. and not layers
but it shows height in mm.
And all the layer information and time information u get from Stefan Cohen's dashboard plugin is broken.. cuz of that..
i have 2 printers runnin the same octopi setup. and the one with v1.15 displaylayerprogress plugin works correctly.. so it's a fault in the 1.16 i guess..

ngroulx, 2020-01-03:
I appear to be experiencing this same issue. With a fresh install of octopi, and v1.16.0 of this plugin.

ngroulx, 2020-01-03:
I was able to confirm that downgrading to v1.15.0 fixed this issue for me.

maz323f, 2020-01-03:
> I was able to confirm that downgrading to v1.15.0 fixed this issue for me.

not sure how to downgrade.. how did u do that ???

ngroulx, 2020-01-03:
> > I was able to confirm that downgrading to v1.15.0 fixed this issue for me.
> 
> not sure how to downgrade.. how did u do that ???

Go to -> Code -> Releases 
- Scroll down to the release version you need.
- Expand assets
-- Either download Master.zip or right-click and "copy link address".

In Octoprint go to -> Settings -> Plugin Manager -> Get More
- Uninstall the current version of the plugin
- Either paste the URL into "...from URL" and click install, or browse to the file you downloaded under "...from an uploaded archive" and click install.

maz323f, 2020-01-03:
ty..  now it working like intended.. ty..

arunsenau, 2020-01-04:
Confirmed this version has broken a few things from what I can see:

1. Layer number ( current_layer ) is broken
2. Model height (total_height) is broken

As per image below.
[image]

OllisGit, 2020-01-04:
Hi @maz323f @ngroulx @arunsenau ,
I am sorry that the new version is not working in your environment.

To analyse the issue I need more informations:
- please attach the gcode provided via download link from octoprint AFTER you select the file. The file must include M117 layer indicator commands (see #104 for more details)
- to analyse the height bug, please make a screenshot of the plugin height settings and attach the g-code as well. I need to know what kind of "height method" do you use.

OllisGit, 2020-01-05:
Hi @maz323f @ngroulx @arunsenau ,
additional question: 

How do you start a print job?
1. via file selection and after that press the print button
2. with small button "Load and print" on the right next to each file

This information would be really helpfull.

tempus2016, 2020-01-06:
I too have this issue.  Height and layer number missing.  I slice in prusaslicer then send gcode to my raspberry pi and select print after sending code.

eiannone, 2020-01-06:
Same problem for me. I upload the gcode from Prusaslicer with autostart. Downgraded to v 1.15 and it's working again.

psu1989, 2020-01-06:
> Hi @maz323f @ngroulx @arunsenau ,
> I am sorry that the new version is not working in your environment.
>
> To analyse the issue I need more informations:
>
> * please attach the gcode provided via download [...]

Here is the beginning of the Gcode.  I don't have a screenshot since i already downgraded to v1.15

;FLAVOR:Marlin
;TIME:10187
;Filament used: 10.0812m
;Layer height: 0.2
;MINX:14.913
;MINY:24.559
;MINZ:0.2
;MAXX:142.58
;MAXY:112.679
;MAXZ:24.6
;Generated with Cura_SteamEngine 4.4.0
M140 S60
M105
M190 S60
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode

G92 E0
G92 E0
G1 F3600 E-1.5
;LAYER_COUNT:123
;LAYER:0
M117 INDICATOR-Layer0
M106 S63.8
M204 S1000
M205 X20 Y20
G0 F685.7 X16.299 Y25.145 Z0.2
M204 S900
M205 X10 Y10
;TYPE:SKIRT

maz323f, 2020-01-07:
i have downgraded to 1.15 and i'm not going to upgrade again.. but in my v1.15 i have 
;LAYER_COUNT:100
;LAYER:0
M117 INDICATOR-Layer0

> Here is the beginning of the Gcode. I don't have a screenshot since i already downgraded to v1.15
>
> ;FLAVOR:Marlin
> ;TIME:10187
> ;Filament used: 10.0812m
> ;Layer height: 0.2
> ;MINX:14.913
> ;MINY:24.559 [...]

maz323f, 2020-01-07:
when uploading from cura the prints start right away..

> Hi @maz323f @ngroulx @arunsenau ,
> additional question:
> 
> How do you start a print job?
> 
> 1. via file selection and after that press the print button
> 2. with small button "Load and print" on the right next to each file
> 
> This information would be really helpfull.

planetar, 2020-01-08:
downgraded to 1.15, thanks to  ngroulx  for the steps!
Everything works nicely again.
Thanks to OllisGit for the plugin and all the work it takes. The readme spells it out clearly, my fault that I read it only after I started the upgrade.
Again, thanks to everyone involved.

Drsmed, 2020-01-12:
> 
> 
> Hi @maz323f @ngroulx @arunsenau ,
> additional question:
> 
> How do you start a print job?
> 
>     1. via file selection and after that press the print button
> 
>     2. with small button "Load and print" on the right next to each file
> 
> 
> This information would be really helpfull.

Hello,
I can confirm that the problem occurs when the "Litle button" next to the uploaded gcode file is used.
When print is loaded in state section, and then I use print button in this section the layer counter works fine.

I use Slic3r with custom code afte each layer change:

;LAYER:[layer_num]

Hope this helps.

OllisGit, 2020-01-19:
Hi @maz323f, @ngroulx, @arunsenau, @tempus2016, @eiannone, @psu1989, @planetar, @Drsmed,   

all mentioned problems should be solved in new version 1.17.0. 
See release news for details: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.17.0

Please test and give me a feedback. 

If you find a bug, please create a new issue. Because in this issue a lot of different things were raised.

Thx, in advance
Olli

arunsenau, 2020-01-20:
@OllisGit awesome thank you! I just tested 1.17.0 and it looks to be working as before now.

Thanks again for all your amazing work on this plugin!

closed 2020-01-28 by OllisGit