OllisGit/OctoPrint-DisplayLayerProgress #107: V1.16&V1.17  height calculation issue

GFernet, 2020-01-21:
Hello since V1.16 and 1.17 the plugin does not know the maximal height of a print and show 0.0mm
[image]
I use a prusa MK3s printer, and check the maximum height using Max Z-Value with Extrusion on Z-Layer.

Is there any change to calculate the height between this two version and the previous one V1.15 which work great.

Thanks

OllisGit, 2020-01-21:
Hi @GFernet,
please attach you gcode (downloaded from OctoPrint) to this issue, so that I can try to reproduce the error.
Thx, in advance
Olli

GFernet, 2020-01-21:
Hi,
My GCode below, but it is the same issue on different files.
[Validation prototype force - Partie fixe_0.2mm_PLA_MK3S_2h41m.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4090179/Validation.prototype.force.-.Partie.fixe_0.2mm_PLA_MK3S_2h41m.gcode.zip)

OllisGit, 2020-01-21:
Hi @GFernet ,

short info: I can reproduce the error with your gcode. I will start analysing asap.

9anSan5, 2020-01-22:
I don't know if my problem is correlated to this one because i noticed it only with last 1.117 version. With last version octoprint ignore a lot of gcode start code of my file sliced with cura 4.4.1. The real problem is that now starts printing from an incorrect (too high) z offset.

OllisGit, 2020-01-26:
Hi @9anSan5,
thats strange!

Please provide the gcode file from cura (save to file) and also the uploaded gcode from octoprint (use download-button i octoprint next to the filename).

Thx, in advance
Olli

OllisGit, 2020-01-28:
Hi all,
I could identified the issue and it will be fixed in next release.

OllisGit, 2020-01-30:
Hi @GFernet, @9anSan5 
new Version 1.17.1 is out. 

Please test and give feedback!

Thx, in advance
Olli

9anSan5, 2020-01-31:
Perfect, now everything is working. Thanks

closed 2020-02-01 by OllisGit