OllisGit/OctoPrint-DisplayLayerProgress #108: [Enhancement] Add local timezone to "estimated_end_time"

OllisGit, 2020-01-21:
As the title says :-)
Add maybe a test/preview-function in Plugin-Settings.

closed 2020-01-30 by OllisGit

brandonscholet, 2020-02-01:
This isn't working for me in the current release and I'm not seeing a setting to adjust.

OllisGit, 2020-02-02:
Hi @brandonscholet ,
please give me more details about "not working"!
I changed the implementation, so that the timezone is picked from the os-system. User settings is not needed. 
Check the clock/timezone on your octoprint-server-host (e.g. raspi).

Because this is a closed enhancement-issue, please raise a new bug-issue or continue in this issue https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/105

brandonscholet, 2020-02-02:
You were exactly right. Sorry to have bothered you. Keep up the amazing work!

Wob76, 2020-02-03:
I have update to 1.17.1 and haven't seen any change, 

I tried both Chrome and Edge, 
I also rebooted my Pi and also cleared the browser cache, am I missing something simple?

The Picture is in the Dashboard, the Msg: lines ETA is from DisplayLayer (13:02), the correct ETA is shown in near middle clock icon "00:01:57", Just past midnight.
[image]

brandonscholet, 2020-02-03:
You have the same problem I have. Your time zone is not correct on the pi. You need to say and copy your timezone over /etc/localtime. This immediately fixed mine.

https://linuxacademy.com/blog/linux/changing-the-time-zone-in-linux-command-line/
https://m.wikihow.com/Change-the-Timezone-in-Linux

Wob76, 2020-02-03:
You're correct, my /etc/localtime was pointing to Europe/London, must be the default, I have updated the symlink to my actual TZ but it is still showing, might just need a new print job to start or maybe a reboot of the pi, I'll update when I reboot and start my next print.

brandonscholet, 2020-02-03:
A new print might work, but I just rebooted octoprint. GMT is Zulu time. All timezones are compared to that.

Wob76, 2020-02-03:
I just tried both tzselect and running raspi-config and still no change, so I think I just need to wait for a new printer\power cycle :)

brandonscholet, 2020-02-04:
Nice