OllisGit/OctoPrint-DisplayLayerProgress #109: Missing total height and estimated total print time in Dashboard

ochm, 2020-01-22:
Hi i updated DisplayLayerProgress Plugin to 1.17.0,
in Dashboard it seems better than 1.16.0, but not all informations are displayed.

[image]

Is again problem in DisplayLayerProgress plugin, or in Dashboard now?
In DisplayLayerProgress Plugin 1.15.0 was everything OK.

ochm, 2020-01-26:
The estimated total print time is displayed only when print starts from Octoprint, not from API (PrusaSlicer).

OllisGit, 2020-01-26:
Hi @ochm ,

a fix for the height bug is currently in progress.
What "total height methode" do you use, see DLP-Settings? 

Regarding the "print time" in the dashboard, this is not a value provided by my plugin.
I looked into the dashboard sourcecode, the implementation just grabs the value from octoprint and displays it...no idea why it is not shown.

Please raise a ticket in the DashBoard-GitHub: https://github.com/StefanCohen/OctoPrint-Dashboard/issues

OllisGit, 2020-01-30:
Hi @ochm, 
new Version 1.17.1 is out. The "total height" is fixed.

Please test and give feedback!

Thx, in advance
Olli

ochm, 2020-02-01:
Yes, Total Height in Dashboard is now OK.
Thank you!

closed 2020-02-02 by OllisGit

HaLo2FrEeEk, 2021-06-26:
Not sure if it's cool to post in this old topic, but I'm having this issue now. I get a - for both total_height and estimated total print time. I know only total_height is under this plugin though. It's a small file, I know that it's 2.5 mm tall, and current_height updates normally, I just never get total_height:

[image]