OllisGit/OctoPrint-DisplayLayerProgress #110: 'DisplaylayerprogressPlugin' object has no attribute '_currentHeightFormatted'

UnchartedBull, 2020-01-23:
If I start the Pi for the first couple of minutes the API always returns a 500 ErrorCode. If I open the UI or wait some time the error usually goes away. Running the latest version (V1.17.0).  Maybe a check, whether self._currentHeightFormatted is defined and returning a default value otherwise might work.

Trace:
```
2020-01-23 17:30:55,254 - octoprint - ERROR - Exception on /plugin/DisplayLayerProgress/values [GET]
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 742, in get_displayLayerProgressValues
    "currentFormatted": self._currentHeightFormatted
AttributeError: 'DisplaylayerprogressPlugin' object has no attribute '_currentHeightFormatted'
```

OllisGit, 2020-01-26:
Hi @UnchartedBull ,
well...testing was not good this time, sorry!

Will be fixed in the next version.
Olli

UnchartedBull, 2020-01-29:
Yeah sure no problem, testing is always tedious and I always forget something to haha. Thanks for including the fix in the next release!

OllisGit, 2020-01-30:
Hi @UnchartedBull ,
new Version 1.17.1 is out. The typo is fixed.

Please test and give feedback!

Thx, in advance
Olli

UnchartedBull, 2020-01-31:
Everything seems to work fine, thanks for the quick fix!

closed 2020-01-31 by UnchartedBull