OllisGit/OctoPrint-DisplayLayerProgress #111: Printing to high after latest update.

82280zx, 2020-01-23:
I have a Ender 3 Pro with TH3D firmware on it, if I have DisplayLayerProgress enabled my printer will not lower to it's z offset when it go's to print it stays at home coords. I have narrowed the issue down to this plugin :(.

9anSan5, 2020-01-24:
Same problem Here. Geeetech A10, Marlin 2.0.x

82280zx, 2020-01-25:
It might be related to the Auto Bed Leveling systems? I had a BLTouch on mine but recently removed it and went back to manual leveling for now and now I don't have the issue.

9anSan5, 2020-01-25:
Could be. I also have the 3d touch on my A10 but i can't try to remove it.

OllisGit, 2020-01-25:
Hi @82280zx, @9anSan5 ,
thats really strange.

Could you please attach a `"octoprint.log"` and the `"plugin_DisplayLayerProgress_events.log"` file to this issue (turn on debugging from DLP Settings, turn off if not needed).

The logfiles should include the following steps:
- starting octoprint
- leveling steps
- selecting file
- start printing

82280zx, 2020-01-26:
I'm not able to reproduce the bug, it happened when I updated this plugin, since then I have reinstalled octopi and things are working as intended.

9anSan5, 2020-01-31:
sorry, i read only now. But with new update everything is working

closed 2020-02-04 by OllisGit