OllisGit/OctoPrint-DisplayLayerProgress #112: Plugin installs but wont "enable"

Andy-ABTec, 2020-01-29:
Hi @OllisGit, apologies, should have gone this way to start with...

I'm trying to use the plugin on a fresh install of:
OctoPrint version : 1.4.0rc4
OctoPi version    : 0.17.0

It installs from the URL quite happily - viz:

    ....
    Successfully built DisplayLayerProgress
    Installing collected packages: DisplayLayerProgress
    Successfully installed DisplayLayerProgress-1.17.0
    Done!

And asks for a re-boot but after rebooting booting  it is disabled, even manually enabling and rebooting doesn't fix it.

octoprint.log has this to say:

    ....
    2020-01-29 07:06:38,443 - octoprint.plugin.core - ERROR - Exception while initializing plugin DisplayLayerProgress, disabling it
    Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1439, in initialize_implementation
        implementation.initialize()
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 225, in initialize
        self._evaluatePrinterMessagePattern()
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 658, in _evaluatePrinterMessagePattern
        if PROGRESS_KEYWORD_EXPRESSION in printerMessagePattern:
    TypeError: argument of type 'NoneType' is not iterable
    2020-01-29 07:06:38,690 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
    2020-01-29 07:06:39,793 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
    2020-01-29 07:06:40,880 - octoprint.util.pip - INFO - Version of pip is 19.3.1
    2020-01-29 07:06:40,883 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
    2020-01-29 07:06:40,884 - octoprint.util.pip - INFO - ==> pip ok -> yes
    2020-01-29 07:06:40,889 - octoprint.plugin.core - INFO - Initialized 14 plugin implementation(s)
    2020-01-29 07:06:40,902 - octoprint.plugin.core - INFO - 15 plugin(s) registered with the system:
      Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
    ....

Not sure where to go from here...

Cheers,

Andy B.

OllisGit, 2020-01-29:
Hi @Andy-ABTec ,
I was able to reproduce your issue. It looks like that you remove the printer message output pattern (default: ```[progress]% L=[current_layer]/[total_layers]```).

Please take a look into your ```config.yaml``` file.
To find out the location of the file, use the folder menu in the settings. The file is located in the parent-folder of e.g. ```uploads```

If ```printerDisplayMessagePattern:``` is empty in your file (see following screenshot), then remove the whole line and restart your octoprint-server.
 
[image]

That should solve the issue.
Olli

Andy-ABTec, 2020-01-29:
Thanks for that, deleted the line and the problem went away...

Content of config.yaml was:

    plugins:
    DisplayLayerProgress: null

Don't understand how this got there as this was a clean install...

Anyway, thanks for the help!

Andy B.

closed 2020-01-30 by OllisGit