OllisGit/OctoPrint-DisplayLayerProgress #117: Error message from DisplayLayerProgress-Settings

freddydelange, 2020-02-15:
Every time I want to print something with Octoprint I get the following error message:
Layer indicator not found in file! Check layer pattern in DisplayLayerProgress-Settings and reupload the file!
The slicer that I use is Cura 4.4.1 but other slicers generate the same message.
My knowledge is not sufficient enough to solve this. Is there someone who can help me with the right settings?
Windows 10
Printrer Creality Ender 3
Cura 4.4.1
Octoprint version Version 1.3.12 with the latest plugins.
 I am almost certain that it must be something in octoprint or the plugin because it also gets me the message with other slicers. am a newbey so it is rather difficult for me maybe it will be easier by e-mail? thanks in advance otherwise don't know how to make a ticket (sorry)

Start and stop Gcodes i use in Cura are:

Cura Start Gcode

G28 ; home all axes
M117 Purge extruder
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X0.1 Y20 Z0.3 F5000.0 ; move to start-line position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
M117 All systems go!

Cura Stop Gcode

G91 ;Relative positionning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning
G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z

And some Sreenshots of DisplayLayerProgress-Settings 
[image]
[image]
[image]
[image]
[image]

Please i need help with this

OllisGit, 2020-02-15:
Hi @freddydelange ,
please do the following:
- Save the generated gcode from Cura to your hard drive
- Open the file with a text editor and look for ```;LAYER:```. Should appear for each layer. If it is not included, it is a slicer issue. Maybe there is a switch for removing all comments (line starting with ;)

- Upload the gcode to OctoPrint
- Download the gcode from OctoPrint to your hard drive (small icon in filemanager)
- Open the file in text editor. Behind each above layer-comment you should see ```M117 INDICATOR-Layer```

Hope that helps!

freddydelange, 2020-02-15:
Hey,
I have gone through the indicated points and I think the Gcode from Cura is good but in the Gcode downloaded from Octoprint I cannot find the M117 I will send you the two files for review.
I like to hear from you

[FromOctoprint_calibration_cube.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4208780/FromOctoprint_calibration_cube.zip)
[FromCura calibration_cube.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4208781/FromCura.calibration_cube.zip)

gdombiak, 2020-02-16:
I'm using Python 3, latest version of this plugin, latest version of OctoPrint (github) and PrusaSlicer 2.1.1. I'm seeing the same error message and found the reason for my issue.

Screenshot:
[image]

Gcode file:
[OUT_E_PETG tempTower_220_235.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4209154/OUT_E_PETG.tempTower_220_235.gcode.zip)

Problem:
In LayerDetectorFileProcessor line 134, the variable **line** is a byte and not a string so _if (len(line) != 0 and line[0] == ";"):_ always returns false.

The following change fixes the problem:

>       if (len(line) != 0):
            line = line.decode('utf-8')
            if (line[0] == ";"):
                for layerExpression in self._allLayerExpressions:
                    inputLine = line
                    line = self._modifyLineIfLayerComment(inputLine, layerExpression)
                    if line is not inputLine:
                        # pattern matched, skip other expressions
                        break
                line = line.encode('utf-8')
            else:
                line = origLine
        else:
            line = origLine
        return line

I have not tested this with previous version of OctoPrint or Python.

OllisGit, 2020-02-16:
Hi @gdombiak , @freddydelange ,

I retested with python 3 and I can confirm that OctoPrint now sends "bin" instead of "str" line-type.
@gdombiak thanks a lot for analysing the issue!

@freddydelange just to make sure: what python version do you use

The issue will be fixed in next version.

OllisGit, 2020-02-16:
Hi @gdombiak ,

you should improve your slicer layer-hook script (if you use the default DLP layer-settings)
Because your gcode includes two comment-indicators: 
```
;BEFORE_LAYER_CHANGE
...
;LAYER:23
```
The result after uploading includes two M117 commands with different layer numbers:
```
G1 X129.165 Y106.594 E0.02765
; stop printing object tempTower_220_235.stl id:0 copy 0
;BEFORE_LAYER_CHANGE
M117 INDICATOR-Layer24
; G92 E0.0
;4.8
;LAYER:23
M117 INDICATOR-Layer23
```

gdombiak, 2020-02-16:
Thanks @OllisGit for catching that. I removed _;LAYER:[layer_num]_ that was added for my temp tower. I now have this as custom gcode before layer change:
>;BEFORE_LAYER_CHANGE
; G92 E0.0
;[layer_z]

It seems like PrusaSlcier does not have a variable that I could use for ;MAXZ: so I guess your plugin will do its best to estimate the max z height.

gdombiak, 2020-02-16:
@OllisGit I opened https://github.com/foosel/OctoPrint/issues/3454 to see if we can do something in OctoPrint so plugins like this one do not break when running with Python 3

OllisGit, 2020-02-17:
Hi @freddydelange , @gdombiak ,
new Version 1.18.0 is out. 

Now it works with:
- OP 1.4.x with Python2/3
- OP 1.13.10 with Python2

Please test and give feedback.
Thx, in advance
Olli

gdombiak, 2020-02-17:
will update tonight! thanks for the super fast turnaround

freddydelange, 2020-02-18:
Hi there I will update tonight too thanks !

freddydelange, 2020-02-19:
Hello all
Unfortunately, after the update to version 1.18.0, the error message is still present.
But thank you so much for all your commitment so far
Again my knowledge level is from a beginner I see several sugestions pass by but would not know where to start,
Kind regards Freddy

OllisGit, 2020-02-19:
Hi @freddydelange,
after installing the new plugin-version, did you re-upload your gcode files again?
Because the layer detection is only done via upload-process. 
So, if you just click on already uploaded files, then you will see the popup.

In a previous version I changed that behaviour (file selection), but then I have a lot of other issues (large-files, autostart, etc.)

freddydelange, 2020-02-19:
Hey,
yes I have reloaded the files and even created new ones to make sure that it was not due to it but it did not help

OllisGit, 2020-02-19:
I think it has nothing to do with Cura. 

Which version of python do you use?

Please look into the octoprint.log file after uploading a new file. Or even better attach the logfile to this issue, maybe I can identify some errors.

freddydelange, 2020-02-21:
Hi there,
Sorry for the late reply I am sending 2 log files
[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4236445/octoprint.log)
 for review and I hardly dare to ask but where do I get the python version information from?
Thanks again and exusses for my ignorance
[octoprint (2).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4236440/octoprint.2.log)

OllisGit, 2020-02-22:
Hi @freddydelange ,
no problem, take your time.

I looked into your two log files:

1) log file 09.02.2020 - 10.02.2020
You use DLP 1.17.1 and often the server was started in SAFE-Mode
```
INFO - Starting in SAFE MODE. Third party plugins will be disabled!
INFO - Reason for safe mode: setting in config.yaml
```
So, DLP will not work!

2) log file 20.02.2020 - 21.02.2020
New Version 1.18.0 is installed, great!
DLP is working and detect the missing layer indicators.
But I can't see an upload event of your selected gcode-file: ```CE3_Woman_body_for_vase_mode_v16.gcode```
So, no fresh upload, no layer detection!

Please do the following
- go to the logging section. end delete octoprint.log
- activate log-level ```debug``` for ```octoprint```
- save settings
- re-upload your gcode file
- select the file and start printing...and stop it
- download ```octoprint.log``` and attach it to this issue

drdeimos, 2020-02-25:
Hi, OllisGit.
It's not working if I uploaded file in directory which is defined in the settings: `Folder -> Upload Folder`?

When do it, doesn't see any messages in `octoprint.log` about my file from DisplayLayerProgress

1larity, 2020-03-01:
This breaks Marlin's K factor test. It is not necessary to have any layer markers for a single layer print.

OllisGit, 2020-03-01:
Hi @drdeimos,
you need to use the browser ui for uploading the gcode (drag&drop). Direct upload to the folder via ssh,ftp doesn't work.
The plugin use the OctorPrint Upload-Event-Listener.

drdeimos, 2020-03-01:
Sad to hear that, but thanks.

drdeimos, 2020-03-01:
I'm not sure for this, but what about to replace upload handling on file handling?
https://docs.octoprint.org/en/1.3.12/events/index.html#file-handling

OllisGit, 2020-03-01:
Hi @drdeimos ,
I already tried this in Version 1.16.0, but then I have a lot of other issues and "race-conditions". With the REST-API for uploading and autostarting the print-job. I received a lot of tickets, so I switched back to my "bullet-proofed" approach.
Maybe I will try it again in later version.

drdeimos, 2020-03-01:
Ok. Thx.

freddydelange, 2020-03-08:
>    1. log file 20.02.2020 - 21.02.2020

Hi there it's bin a while but here it is the requested log file made today
Before I printed this I updated to the latest version of the plug-in but I
get the same message as before. So I hope you can help me with this issue

Thanx

OllisGit, 2020-04-05:
Hi @freddydelange ,
there is no log-file attached!
Don't use the Email-Answer function. Doesn't work with attachments.
You need to upload the file via GitHub-UI.

OllisGit, 2020-04-08:
Hi,
new version 1.19.0 is out now and solves some issues.
@drdeimos you can now disable adding Layer-Indicator-Command during upload, then you can put this command into your slicer. See https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.19.0

closed 2020-08-15 by stale[bot]