OllisGit/OctoPrint-DisplayLayerProgress #119: Argument of 'NoneType' is not iterable

ciordia9, 2020-02-18:
[octoprint-1.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4221809/octoprint-1.log)

As requested, removed DisplayLayerProgress, Octoprint logs in debug, reinstall, reboot.

`2020-02-18 16:46:28,873 - octoprint.plugin.core - ERROR - Exception while initializing plugin DisplayLayerProgress, disabling it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1439, in initialize_implementation
    implementation.initialize()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 245, in initialize
    self._evaluatePrinterMessagePattern()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 732, in _evaluatePrinterMessagePattern
    if PROGRESS_KEYWORD_EXPRESSION in printerMessagePattern:
TypeError: argument of type 'NoneType' is not iterable`

OllisGit, 2020-02-19:
Hi @ciordia9 ,

looks like issue #112. There is something wrong with the dlp plugin settings. 
Please do the advice from the mentioned ticket.
At the moment I have no clue why the settings could be broken, because this is a "standard feature" of OctoPrint.

ciordia9, 2020-02-25:
Hiya @OllisGit, sorry for the delay, like #112 you linked to I had the same :null line, removing that and everything here goes back to normal.

If you need anything else from me to test let me know.

closed 2020-03-01 by OllisGit