OllisGit/OctoPrint-DisplayLayerProgress #120: Help , Any guide for add layer-indicator in Simplify3D?

Askeshine, 2020-02-19:
Hi , any have guide for add layer-indicator in Simplify3D i read alot of and i dont have chance...

[image]

GCODE: 

[image]

OllisGit, 2020-02-19:
Hi @Askeshine ,
hmmm...actually you don't need a separate "layer change script', because Simplify3D already create "layer-comments" like this ```; layer 1, Z = 0.100```.

Please look into the DLP plugin-settings / Layer, there must be this entry:
```1		[; layer ([0-9]+),.*]		Simplify3D```

If you could not find an error, please attach your uploaded gcode (directly from slicer + downloaded from octoprint, small button) and also your "layer-expressions" to this issue.

OllisGit, 2020-04-06:
No feedback

closed 2020-04-06 by OllisGit