OllisGit/OctoPrint-DisplayLayerProgress #122: Extruder not heat when plugin active

ldiegos, 2020-03-01:
After installing the plugin the Creality Ender 3 stops extruding. I check that the pid to start heating is not shown on the top horizontal bar, only the heating bed pid

This are the plugins installed on my octoprint, maybe is not compatible with some of them:
Convert TF to SD - not in repository
Dropbox Timelapse Plugin
Exclude Region
Floating nav bar
GcodeEditor
Navbar temperature Plugin
Octolapse
PrintTimeGenius Plugin
Progress Title
PSU Control
TabOrder
Themeify
WebcamTab

OllisGit, 2020-03-01:
Hi @ldiegos ,
I need more information about the issue.

- Please attach the octoprint.log file to this issue
- If you only disable DLP is your printer working normal?

Btw. with DLP V1.7.0+ you can define your own progress title. If you still want to use the "Progress Title"-Plugin, ,then you need to disable the "Browser-Title" checkbox in the Plugin-Settings

ldiegos, 2020-03-01:
Thank you so much for your answer @OllisGit !!!

Yes, if i disable the plugin and restart, as octoprint recommend, everything goes fine.

The log start yesterday at 22:50.
I installed the plugin at 22:54 and then... zzZzzZZz  lol
2020-02-29 22:54:39,235 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/latest/download/master.zip

This afternoon when i tried to start printing some stuff, started the issue and this is the error i found in the log file.

`2020-03-01 18:03:51,698 - octoprint.util.comm - ERROR - Error while processing hook DisplayLayerProgress for phase sending and command M104 S220.0 ï»¿T0:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 280, in sendingGCodeHook
    commandAsString = str(cmd)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufeff' in position 12: ordinal not in range(128)
2020-03-01 18:03:51,777 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': '48a49a22ab00fc9629d9300a46a9c633242f1c4c'}
2020-03-01 18:04:01,661 - octoprint.util.comm - ERROR - Error while processing hook DisplayLayerProgress for phase sending and command M109 S220.0 ï»¿T0:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 280, in sendingGCodeHook
    commandAsString = str(cmd)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufeff' in position 12: ordinal not in range(128)`

Here is the full log.
[octoprint_ldiegoes_20200301.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4272274/octoprint_ldiegoes_20200301.log)

OllisGit, 2020-03-01:
Hi @ldiegos ,
looks like an encoding problem ```UnicodeEncodeError: 'ascii' codec can't encode character```
If possible, please attach your gcode ```Eryone-CE3_AUGMagazine-23gLines10Per.gcode```

ldiegos, 2020-03-01:
Which one? 
The gcode generated by cura 4.4.0? or 
The gcode after uploaded to octoprint that suffer the modifications from all the plugins?

Kind regards, 
Luis.
PS: I'm a rookie  sending issues to github :)

OllisGit, 2020-03-03:
Hi @ldiegos ,

just add both files in a zip to this issue.

tech2077, 2020-03-06:
I also had this identical issue. I found a common Cura Start GCode had an unicode space in it. Here's a file generated that includes this space (verified using: [https://www.soscisurvey.de/tools/view-chars.php](https://www.soscisurvey.de/tools/view-chars.php)).

[unicode.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4300972/unicode.gcode.zip)

OllisGit, 2020-03-08:
Thx @tech2077 for the provided gcode, now I could reproduce and fix the error.
I tested the newest version 1.18.1 against OP1.3.10 and  OP1.4.0 Python2/3.

@tech2077 , @ldiegos , please also test the new version and give me a feedback.

Thx, in advance
Olli

ldiegos, 2020-03-09:
Good morning,
Sorry for the delay, my last week was a little chaotic. I attached both files. 

I will test your new version as soon as possible.

[Eryone-CE3_AUGMagazine-23gLines10Per.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4304788/Eryone-CE3_AUGMagazine-23gLines10Per.zip)

ldiegos, 2020-03-11:
Hi,
It seems to work right now. I'm sending a new gcode to the CE3 and the extruder pid is working.
I'm going to check if all the functionality remains working.

Thank you so much.
Luis.
[image]

closed 2020-04-06 by OllisGit