OllisGit/OctoPrint-DisplayLayerProgress #123: Exclude folder(s)

ToXIc-Dev, 2020-03-02:
I am using my printer as a makeshift CNC and when using the gcode files for it this plugin constantly gives me this error notification "Layer indicator not found in file! Check layer pattern in DisplayLayerProgress-Settings and reupload the file!"

I have my files in a separate upload folder called CNC so I was wondering if you could add functionality to exclude a certain folder(s) from analysis (or maybe files with a certain prefix, folder would be better IMO though)

OllisGit, 2020-03-03:
Hi @ToXIc-Dev ,

nice idea! 
Will check the effort and come back to you.

OllisGit, 2020-08-17:
Hi @ToXIc-Dev ,
it is implemented in the newest version 1.23.0

Please test and give a feedback

Thx, in advance
Olli

closed 2020-09-26 by stale[bot]