OllisGit/OctoPrint-DisplayLayerProgress #126: Error 500 when uploading a new file after latest update

5jplusa, 2020-03-07:


5jplusa, 2020-03-07:
on the last 2 updates I get error 500 when uploading new files, the only way to correct is to roll back to previous version

OllisGit, 2020-03-07:
Hi @5jplusa ,

what version of OP und DLP do you use? 
What python version do you use?
Please attach the octoprint.log and also the gcode you receive the 500er error.

Olli

5jplusa, 2020-03-18:
OctoPrint 1.4.0 running on OctoPi 0.16.0

[octoprint.log.2020-03-16.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4347740/octoprint.log.2020-03-16.log)

5jplusa, 2020-03-18:
currently using DisplayLayerProgress Plugin (1.17.1) and upload is working perfect.

5jplusa, 2020-03-18:
[Filament Guide - Gold.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4347791/Filament.Guide.-.Gold.gcode.zip)
here is a sample Gcode that would cause the error

OllisGit, 2020-03-18:
Hi @5jplusa ,

I am a little bit confused: Is it's now working with 1.17.1? 
btw. current Version 1.18.1
I can't see any errors in the logfile that is affected to DLP.

5jplusa, 2020-03-20:
Do you want me to reinstall 1.18.1 and send you the log file after trying to upload a file?

5jplusa, 2020-03-20:
updated to latest version and still cannot upload. attached are the log files and the gcode 
[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4358442/plugin_DisplayLayerProgress_events.log)
[octoprint (3).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4358443/octoprint.3.log)
[Filament Guide Wire Keeper.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4358444/Filament.Guide.Wire.Keeper.gcode.zip)

5jplusa, 2020-03-20:
reverted back to 1.17.1 and it is working again, attached are the logs after reverting back to 1.17.1
[octoprint (4).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4358565/octoprint.4.log)
[plugin_DisplayLayerProgress_events (1).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4358566/plugin_DisplayLayerProgress_events.1.log)

OllisGit, 2020-04-08:
Hi @5jplusa, 
it is fixed in the newest release V1.19.0! Was a unicode-issue.

Please test and give feedback.

Thx, in advance
Olli

gmag11, 2020-04-09:
Hello, I'm getting same error with 1.19.0 with OctoPrint 1.4.0
[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4457270/octoprint.log)
[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4457271/plugin_DisplayLayerProgress_events.log)

OllisGit, 2020-04-10:
Hi @gmag11,
I could reproduce the error. It is related to Python 3.

I think you also use some messages that use not "normal characters" (sorry to say, I have no other translation for that).
Like
```
b'M117 Priming Filament\xe2{\xa6\n'
```
To verify this, please attach your gcode that throws the 500er error.

Thx, in advance
Olli

OllisGit, 2020-04-10:
@5jplusa question: Are you "Korsisch-Speaking" ? 
Because if I use google translate for ```M117 Priming Filamentâ``` the result is Korsich for Filament

ax42, 2020-04-10:
Same issue here - Octoprint 1.4.0, Python 3.6.  Disabling the plugin allows normal operation.  Uploading from PrusaSlicer or uploading from the OctoPrint UI both give error 500.

OllisGit, 2020-04-10:
Hi @ax42,
please attach your gcode so that I am able to reproduce the error.

ax42, 2020-04-10:
Fails on any file.  Here is one.

[jst_0.15mm_PLA_MK3S_3m.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4462045/jst_0.15mm_PLA_MK3S_3m.gcode.zip)

OllisGit, 2020-04-10:
Hi @ax42 ,
thanks for the gcode, this was really helpful!

I just released version 1.19.1 this should fix the error. I tested with Py2/3.

Please test and give me a feedback!

ax42, 2020-04-11:
Worked for me, thanks.
Moot now, there was something I hadn't written here -- I didn't need to disable the plugin to get rid of the error message, I only needed to unclick the "add clayer indicators during upload".

closed 2020-04-18 by OllisGit