OllisGit/OctoPrint-DisplayLayerProgress #128: Communication timeout after N38/N39 with Octoprint 1.4.0 and DisplayLayerProgress version 1.18.1

Kelly12612, 2020-03-26:
My printer is a Prusa MK3s firmware 3.8.1-2869 with MMU2s firmware 1.0.6-372.
After upgrading to octoprint 1.4.0, my first two prints aborted with a communication error.
I did some testing and found that the same thing happened if I just uploaded a gcode file to the SD card.
I did a lot more testing and found that the problem only occurs if the DisplayLayerPlugin is enabled, if I disable that, everything works, even with all my other plugins enabled.

It happens with gcode files from PrusaSlicer versions 2.1.1 and 2.2.0 (ga).

I've attached a terminal log from a failed send, and the octoprint.log file from a failed send.

The problem always seems to occur after an N38 or N39 command, one example is:
Recv: ok
Send: N39 G92 E0.0*99
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

Let me know if there is any other information you'd like, it's pretty easy to reproduce.
[terminal_log_march_26_0940.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4387891/terminal_log_march_26_0940.txt)
[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4387892/octoprint.log)

OllisGit, 2020-04-03:
Hi @Kelly12612,

which Python-Version do you use?
If possible, please attach the original sliced g-code from the slicer and also the upload g-code downloaded from OctoPrint-Menu.

Kelly12612, 2020-04-03:
This was an upgrade, so it appears to still be running the older python:
pi@octopi:~ $ python -V
Python 2.7.16

Kelly12612, 2020-04-03:
[dlp_128_original.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4430132/dlp_128_original.txt)
[dlp_128_downloaded_from_octoprint.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4430133/dlp_128_downloaded_from_octoprint.txt)

OllisGit, 2020-04-08:
Hi @Kelly12612 
in the newest release 1.19.0 I changed the behaviour (switching to async) of processing the gcode during print. I think this was your issue, because the gcode was not send fast enough to the printer.

Because it is just an assumption, please test and give me a feedback.

Thx, in advance
Olli

Kelly12612, 2020-04-10:
The problem is still happening for me with the new release. I've attached a smaller gcode file that reproduces the problem.  If I try to copy this to the SD card through Octoprint with this plugin enabled, it gets partway through then I get a communications error/reset (failed 3 times in a row).
If I disable the plugin and restart, the file is transferred successfully.

[SuperLubeTube_and_cap_x3_PLA_0.15mm_PLA_MK3SMMU2S_2h4m.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4462286/SuperLubeTube_and_cap_x3_PLA_0.15mm_PLA_MK3SMMU2S_2h4m.zip)

Kelly12612, 2020-04-18:
I tested it and updated the ticket (still a problem), but forgot to respond
to this email until now.
-Kelly

OllisGit, 2020-04-18:
To be honest, currently I have no idea how to analyse this....but I want to mention one point: local vs sd card-upload

Local
upload goes to the pi. OctoPrint can listen on gcode and could adjust it (like DLP)
SD
goes over the serial connection to the SD card in the printer. During the print OctoPrint can't listen/manipulate gcode, because it is directly send to the print (DLP don't work). Datatransfer is faster.
More information, see https://community.octoprint.org/t/ouch-sd-card-vs-octoprint-what-a-difference-help/8618

OllisGit, 2020-04-18:
One idea: I looked into your Serial log again and some printers had issues with "special-characters" for the M117 display command. 
Try to send the following commands to your printer via terminal to reproduce the error.
```
M117 INDICATOR-Layer1
``` 
and
```
G92 E0.0
```
Maybe the printer can't handle ```-``` and the next command could not be send.....just an idea!

Kelly12612, 2020-04-19:
No problem with those commands.
-Kelly

OllisGit, 2020-04-24:
Hi @Kelly12612, 

last idea: Try latest development version https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.20.1dev/master.zip

This version use queing-hook and is not interfering the communication between OP and Printer.
The values in the display are now a little bit behind.

Minglarn, 2020-04-24:
How to update to 1.20.1dev from octoprint?

Kelly12612, 2020-04-25:
Still no joy, I'm afraid, failed while copying to SD, log below, version
displayed in plugin manager
was:  DisplayLayerProgress Plugin (1.20.1.dev0)

Send: N39 G92 E0.0*99Recv: okSend: N40 G1 E-0.80000
F2100.00000*9Communication timeout while printing, trying to trigger
response from printer. Configure long running commands or increase
communication timeout if that happens regularly on specific commands
or long moves.Send: N41 M105*18Communication timeout while printing,
trying to trigger response from printer. Configure long running
commands or increase communication timeout if that happens regularly
on specific commands or long moves.Send: N42 M105*17Communication
timeout while printing, trying to trigger response from printer.
Configure long running commands or increase communication timeout if
that happens regularly on specific commands or long moves.Send: N43
M105*16Communication timeout while printing, trying to trigger
response from printer. Configure long running commands or increase
communication timeout if that happens regularly on specific commands
or long moves.Send: N44 M105*23Communication timeout while printing,
trying to trigger response from printer. Configure long running
commands or increase communication timeout if that happens regularly
on specific commands or long moves.Send: N45 M105*22No response from
printer after 6 consecutive communication timeouts, considering it
dead. Configure long running commands or increase communication
timeout if that happens regularly on specific commands or long
moves.Changing monitoring state from "Sending file to SD" to "Offline
(Error: Too many consecutive timeouts, printer still connected and
alive?)"Connection closed, closing down monitor

OllisGit, 2020-04-25:
@Minglarn  use the plugin-manager http://docs.octoprint.org/en/master/bundledplugins/pluginmanager.html and enter the above master.zip-url into "from URL" and press install.

OllisGit, 2020-04-25:
Hi @Kelly12612 ,

hmm..short Summary: Two problems
1. Upload to printers SD-Card is not possible, if plugin enabled 
2. Sending gcode during print from OP-RasPI-FlashCard throws communication-error.

Lets focus on issue 1)
- You upload the gcode via Browser-UI und choose SD-Card (Drag&Drop right-side), right?
- What error do you receive if upload failed? Please attach the OctoPrint-Log.

Kelly12612, 2020-04-26:
[dlp_issue_128.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4534484/dlp_issue_128.zip)

Kelly12612, 2020-04-26:
Sorry, I guess I wasn't clear, all the logs I've uploaded have been about
the upload issue.
Yes, I drag the file to the right side of the octoprint web page.
I've uploaded to the ticket, a zip file that has logs from uploading the
same file, with and without DisplayLayerPlugin.
-Kelly

OllisGit, 2020-06-12:
Short status update: 
- I looked into your logs and I am wondering why the printer stops receiving data only after 3 seconds
```
2020-04-25 19:19:11,695	Changing monitoring state from "Operational" to "Sending file to SD"
...
2020-04-25 19:22:12,492 No response from printer after 6 consecutive communication timeouts, considering it dead.
2020-04-25 19:22:12,563 Changing monitoring state from "Sending file to SD" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
```
- Maybe a combination of the "exclude-plugin" raise this issue, because the "exclude" also use a file-preprocessor during the upload

Do me a favour and test uploading files with disabled "exclude-plugin".

Thx, in advance
Olli

closed 2020-08-15 by stale[bot]