OllisGit/OctoPrint-DisplayLayerProgress #130: [Feature Request] Python 3 Compatibility

jneilliii, 2020-03-31:
**Is your feature request related to a problem? Please describe.**
Running the latest version of OctoPrint 1.4.0 in a Python 3 environment plugin is not found in Plugin Manager, because of compatibility flags. I see the plugin is compatible in `__init__.py` but plugin repo has not been updated to list it as compatible.

**Describe the solution you'd like**
Update Plugin Repository with proper compatibility information.

**Additional context**
Please see [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready) for the required updates.

gvmturl, 2020-04-03:
Waiting for that!

OllisGit, 2020-04-03:
Done..and merge by you @jneilliii ;-)

@gvmturl btw. DLP is working with python3 and OP 1.4. since version 1.15.0 (Jan2020), see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/89#issuecomment-568199147
Only the compatibility information was missing in the plugin-repository.

closed 2020-04-03 by OllisGit