OllisGit/OctoPrint-DisplayLayerProgress #132: Edit state

radimkohout, 2020-04-03:
Hi,
is there any way to edit the state tab on the left? There are current height and layer added by this plugin, but I can't find if I can edit it somewhere

OllisGit, 2020-04-03:
Hi @radimkohout,
currently there is no edit-option, because I don't see the demand on that.

Please explain in detail why you need such functionality.

radimkohout, 2020-04-04:
I need it for displaying [estimated_end_time] and [feedrate] in here.

OllisGit, 2020-04-08:
Hi @radimkohout,
it is implemented in the newest release V1.19.0!

Now you can customise the state-section like the nav-bar.

Please test and give feedback.

Thx, in advance
Olli

radimkohout, 2020-04-08:
It looks, that everything is working flawlessly. 
Thank you

closed 2020-04-08 by radimkohout