OllisGit/OctoPrint-DisplayLayerProgress #133: ;LAYER:[layer_num] not parsing due to whitespace?

pfakanator, 2020-04-05:
My gcode has always injected ;LAYER:[layer_num].  I add white space for better readability when hand editing.  Could this be causing parsing errors.  Example:

G1 X-47.879 Y-49.733 E1563.82761
G1 X-48.055 Y-49.910 E1563.83706
M126
;BEGIN  z-pos:0.6
;LAYER:1

M73 P3
G1 F5760
G1 X-47.879 Y-49.733 E1563.78768
G1 X-47.703 Y-49.910 E1563.73829
G1 E1562.83706 F1200.00000

OllisGit, 2020-04-06:
Hi @pfakanator ,
the next white line shouldn't be a problem.

Please attach a sample g-code so that I can debug the issue in a more deeper.
It is also helpfull when you could put the layer-expressions from the plugin-settings to this ticket (only if you changed it)

pfakanator, 2020-04-07:
Using Slic3r. GCODE layer change injection is as follows:

> 

;BEGIN  z-pos:[layer_z]
;LAYER:[layer_num]

>

Example GCODE:
 [example.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4445401/example.zip)

OllisGit, 2020-04-08:
Hi @pfakanator ,
I couldn't use the example file, my real and virtual printer crashed (also without DLP installation).

Reason: some line starts with ```(**...``` instead of ```;(**...)```.

I attached a modified version and with this version I could see all layer-informations (after file selecting and during printing).

Please test this version.
[example-modifiedByOlli.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4449876/example-modifiedByOlli.gcode.zip)

pfakanator, 2020-04-08:
I loaded up your GCODE and the printer (original replicator2) ran it just as it did w/out your edits.  I still get "Layer indicator not found in file! Check layer pattern in DisplayLayerProgress-Settings and reupload the file!"

OllisGit, 2020-04-08:
@pfakanator hmm...strange..in the newest Version 1.19.0 (released today) I fixed a unicode-issue. Maybe it is also related to your problem.

Please so kind and test the new version and give me feedback.

Thx, in advance
Olli

pfakanator, 2020-04-08:
Still no luck ;(.  Attached is the log.
[plugin_DisplayLayerProgress_events.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4453063/plugin_DisplayLayerProgress_events.zip)

OllisGit, 2020-04-18:
After installing the Plugin, did you reupload the gcode via UI? 
This is necessary, because the LayerIndicator will be generated during UI-upload.

Please try to reupload and if it still not working please attach the gcode file and also the octoprint.log file.

EvilGremlin, 2020-04-18:
What about API uploads? PrintTimeGenius process everything just fine, what DLP don't do that?

pfakanator, 2020-04-18:
Ah ha!  I was exporting from Slic3r directly to octoprint.  When I upload the gcode via the web interface the file parses properly.  It's a change to workflow but hey, it works.  Thanks!

OllisGit, 2020-04-19:
@EvilGremlin I should have written the following: "..during UI-upload or File-API"
E.g. the Cura-Plugin use this API

@pfakanator do you know how Slic3r communicate with OP? Is it just a file-copy? Maybe there is also a Plugin (like in Cura) for communicate with OP.

pfakanator, 2020-04-19:
I know Slic3r requires and API key, however, I don't know how it works under the hood.

OllisGit, 2020-04-24:
Hi @pfakanator ,
if Slic3r uploads the gcode directly to a SD-Card (not PI-SD), then my plugin also not work, because the gcode-file is send directly to the printer. OctoPrint is not able to read each line.

pfakanator, 2020-04-25:
I do not use an SD card in my printer.

OllisGit, 2020-06-04:
Hi @pfakanator,
I just installed slic3r to reproduce the "send to printer" issue. I added your layer-change script and after sending to OP everythings works fine.

Which version of slic3r do you use? I installed Version 1.3.0 on a Mac.

pfakanator, 2020-06-07:
PrusaSlic3r 2

OllisGit, 2020-06-07:
I tested with PrusSlicer 2.2.0+, because I could not find PrusSlic3r 2. It worked fine.
Please provide a link to your software...or is this issue still present for you?
[image]

closed 2020-08-15 by stale[bot]