OllisGit/OctoPrint-DisplayLayerProgress #136: Release 1.19.0 broke OctoPod app

gdombiak, 2020-04-10:
OctoPod used to read **stateMessage** that until now was the current layer info. I also read **heightMessage** for height information. I now see that **stateMessage** is dynamic and includes HTML. Now clients using your plugin in OctoPod get this on their phones

[image]

If you are ok, I can create a PR that introduces a new field called **layerMessage** that will contain the same information we used to have before in stateMessage. If we go this route, things will be broken until users update to the new version of OctoPod and the new version of your plugin.

The other alternative, is to revert **stateMessage** so that it has the same info as before and create a new htmlStateMessage property that is what the UI uses. This option will only require users to update to the new version of this plugin.

What do you think? Any preference?

Thanks,
Gaston

OllisGit, 2020-04-10:
Hi @gdombiak,
I did not know that you "hijack" informations from the DLP-UI.
IMHO: This is not a good approach! The UI is not a stable interface (no API).

You should better use one of the Developer-Interface:
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress#developer---section

- REST API
- Listening on EventBus 

Dashboard-Plugin switched also from view-model to Eventbus (long time ago).

If you have problems how to handle the eventbus stuff I can support you.

C ya,
Olli

gdombiak, 2020-04-10:
Hi @OllisGit,

I was not aware of the event bus so didn't use it! :) I looked at the documentation and it's not clear to me if event bus broadcasts events via websockets or is this local inside of Python for other plugins? 

OctoPod is an iOS app that relies on websockets to get real time information from all supported plugins. Polling through the REST API won't be a good option that is why the question about websockets.

Thanks,
Gaston

gdombiak, 2020-04-10:
Quick break at work. Seems like https://github.com/OctoPrint/OctoPrint/blob/5596daf34668b90fa851ff2b3b0f1730413c7d69/src/octoprint/server/util/sockjs.py#L374 might be sending via websockets the events published to the event bus. It seems to subscribe to all events and then send them via #sendEvent. I will test this once out of work.

Thanks,
Gaston

gdombiak, 2020-04-11:
Not good news. [sockjs.py subscription](https://github.com/OctoPrint/OctoPrint/blob/5596daf34668b90fa851ff2b3b0f1730413c7d69/src/octoprint/server/util/sockjs.py#L374) only subscribes to OctoPrint's native events and not to events coming from plugins. This means that event bus is not broadcasting events via websockets.

FYI, Apple TV app is also broken and people are starting to report this. :( Would you oppose going with option 1 that I mentioned above? It is a common practice for plugins to treat #send_plugin_message as a public API (i.e. public contract and not break backwards compatibility). This is how integration with Pallete2 plugin, CancelObject plugin, TPLink, Tasmota, Wemos and many more plugins was done.

Thanks for your help,
Gaston

OllisGit, 2020-04-11:
Hi @gdombiak,
I have a different opinion on that. For me ```send_plugin_message``` is the way for plugin-server / -ui communication. In my case the payload includes ui specific content, like HTML-Code that is related to CSS/JS-Code.

How should I enhance my plugin (like I did with 1.19.x) with more UI-Features when I am blocked not to change anything in the structure? 
Thats the reason why I establish the eventbus-/rest-api for 3rd Party-Developer. This is stable!

Right now, it is not 100% clear for my, why you need this information from the UI.

I looked into your OctoPrint-Plugin and I could not identify where you grab the layerMessage from the UI. Is it in the swift-code?
But I saw that you already listen to DLP-Events: https://github.com/gdombiak/OctoPrint-OctoPod/blob/bb13222413036417f5f066bc38a9bd465a1c05ba/octoprint_octopod/__init__.py#L142

Please explain why is it not possible to use this approach?

From my current understanding you are "listening" to the messages that are send via "send_plugin_message"...as a compromise I am thinking about to include the whole rest-payload (https://github.com/OllisGit/OctoPrint-DisplayLayerProgress#rest---api) as an additional attribute to the ui-payload. 
Or an other option is that I send the rest-payload with "send_plugin_message" with a second plugin-identifier.

Instead
```
self._plugin_manager.send_plugin_message("DisplayLayerProgress", dict(...))
```
I use
```
self._plugin_manager.send_plugin_message("DisplayLayerProgress-APIModel", dict(...))
```

But I didn't want to increase the UI-Traffic. I will also ask the community what is the best way to do.

gdombiak, 2020-04-11:
Thanks @OllisGit for the response. Maybe this helps explain more the situation. The OctoPrint-OctoPod plugin was created for sending push notifications to the app. Of the over 23K users of OctoPod only a very small percentage (5% ?) use the plugin. The iOS app and the Apple TV app read information directly from your plugin and all the other plugins I mentioned. 

_If the way to solve this problem is to require the OctoPrint-OctoPod plugin then most users that used to see layer information will stop seeing it._

The other plugins that I mentioned do evolve. They keep adding and improving their functionalities. The way they achieve that is by defining the semantic of each variable. If they want to change the semantics then they create a new variable since it is a different semantic. If they want to create a new semantic then they create a new variable. This is why I suggested option 1 that would have a clear semantic for the new variable **layerMessage** just like you do for **heightMessage**. If desired, new variables for html version of those concepts could coexist. 

Let me know how you want to proceed.
Thanks,
Gaston

OllisGit, 2020-04-11:
Hi @OllisGit,

**heightMessage** is a leftover from Version 1.18.x, in the next release it's gone.
Then I only send state/navBarMessage and browserTitle.

Please give me more technical inside how do you receive the layer-information (source-code).
Do you use a Browser-View component in your app and listen on websockets-messages?

I understand your pressure to solve that issue and also not to "force" the user to install an additional plugin.

My planned  solution will looks like this: 
- push a second message with different Plugin-Identifier. The payload will be the event based payload
-   This mechanism needs to be manually activated by the user in the DLP-Settings, to reduce network traffic

gdombiak, 2020-04-11:
OctoPod relies on websockets to get real time information from OctoPrint and installed plugins. For your particular plugin reception and reaction to the data happens [here for iOS](https://github.com/gdombiak/OctoPod/blob/master/OctoPod/Panel%20UI/Subpanels/PrinterSubpanelViewController.swift#L167) and [here for Apple TV](https://github.com/gdombiak/OctoPod/blob/35832463254aa8634751b650e91416a560d20a71/OctoPod%20TV/ViewService.swift#L171).

I think that the idea of having a new plugin-identifier that you treat as public API is a great one. you will have the freedom to do whatever internal changes with the current plugin-identifier without any concerns of breaking other clients.

Here is an idea, in order to make it dead easy to start seeing layer information in other clients like OctoPod I like to reduce setup steps. Install your plugin and install OctoPod and you are good to go. I understand the idea of reducing network traffic that can impact CPU processing and therefore battery lifetime in mobile devices. Looking at [event paylolad](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/blob/c0526a66d93e68477128ca526bbf11dc2c6448c2/octoprint_DisplayLayerProgress/__init__.py#L839) I wonder if we could:
1. Have an event that by default is fired with fields up to _ fanspeed_. Users can opt-out if desired in DLP-Settings
1. Have an event that by default is disabled that includes time left information (things after _fanspeed_). Users can opt-in if desired in DLP-Settings

If splitting things into 2 events will not reduce frequency of sending data then my proposal is to by default send the event with an opt-out option. People that are on desktops won't mind for the extra traffic and people on mobile devices that use OctoPod won't be impacted that bad in battery terms since duration of sessions is not that long.

Thanks again,
Gaston

jr-air, 2020-04-11:
Am glad to see a solution is being considered here. 

Honestly, am sure I am speaking for many, but both Octopod and DisplayLayer are two of the best additions/enhancements to Octoprint. Thanks to both of you for your great work!

OllisGit, 2020-04-11:
Hi @gdombiak,
I pushed a new Version 1.20.0dev. It is available here: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.20.0dev/master.zip

You need to install the plugin via Plugin-Manager and after a restart you need to activate the websocket-feature in the settings (first page in the bottom).

After that, you can grab the messages via this Plugin-Identifier: ```DisplayLayerProgress-websocket-payload```

---
To be honest I don't understand your feature request. What do you mean with "... fields up to _ fanspeed_". 
You want the event-payload (https://github.com/OllisGit/OctoPrint-DisplayLayerProgress#events) should be customizable or at least two typs of attributes (must-have, optional). Is that what you say?

gdombiak, 2020-04-12:
Thanks @OllisGit for the quick work. Much appreciated. I think that we are going in the right direction.

My remaining concern is that most users will not realize that they have to go to settings to enable this feature. They will assume that things are broken since they have the plugin installed and OctoPod is not showing any information. Would you be ok with making **SETTINGS_KEY_SEND_LAYERINFORMATION_VIA_WEBSOCKET** default to True? There won't be any impact if no websocket client is connected. If a browser is connected then it will receive this extra payload and browsers on desktops should be ok with the extra network traffic. Browsers on mobile devices are used for a few minutes so impact would be minimal since TCP connection is already established. And the same applies to OctoPod.

Thanks again,
Gaston

OllisGit, 2020-04-14:
...yes, I can change the default-value to True!
Did you already test Version 1.20.0dev? Is it ready "for production"?

gdombiak, 2020-04-14:
Hi @OllisGit, sorry I forgot to update this thread. The new version works perfect! Enabling this by default will make it easy on people. Thanks much for being open to feedback!

Once you release the new version, I will push the new version of OctoPod so users can see layer info again on their TVs and iPhones.

Thanks,
Gaston

gdombiak, 2020-04-16:
Hi @OllisGit, I'm done implementing all the features for OctoPod 3.3. Let me know when you changed the default to True and released the new version. I'll wait for you to be done before submitting app to store.

Thanks,
Gaston

JaySea77, 2020-04-17:
You asked if I wanted to beta test in the AppStore.  Please tell me how.

gdombiak, 2020-04-17:
Good news @OllisGit. More people are using your 1.20.0de version and the new beta of OctoPod and things are working again. Let me know when you changed the default to True and released the new version. Once you did that I will submit the new version of the app to be officially released.

OllisGit, 2020-04-18:
Hi, @gdombiak ,

new version 1.20.0 is out now!

BR
Olli

gdombiak, 2020-04-18:
Awesome!!! Thanks @OllisGit. Will push new version to the App Store now.

Thanks,
Gaston

closed 2020-04-18 by gdombiak