OllisGit/OctoPrint-DisplayLayerProgress #137: Some general questions about latest version...

Protoncek, 2020-04-13:
Hi!
I'm testing latest version and i have some questions.
I use Prusa Slicer, it's basically pretty same as Slic3r. I also use OctoDash plugin for my 7 inch LCD. I have RPi4-4GB, so there's plenty of computing power.
How do you guys upload g-codes into Octoprint? Where? how? i upload it into "watched" folder. Is this correct? Namely, interesting is that afer i power-up my system this seems to work fine - i slice my file with default profile, upload it into octoprint's "watched" folder and octoprint transfers it into "uploads" folder and i can print it. However, when i start to print something this transfer is no longer happening and g-code stays in "watched" folder. Even when printing is finished file is still NOT transferred. I have to restart octoprint to work again. Am i putting my files into correct folder?

Second: if i uncheck "Add Layer-Indicators during upload" and i put myself "M117 INDICATOR..." info manually and i store file directly into "uploads" folder  plugin still says that file is corrupt (doesn't find layer info), even file is exactly same as if it would be processed. 
So, is file processing mandatory, no matter if plugin adds "M117 INDICATOR..." info or if i add it myself? 

I hope that you understand what i'm trying to ask, since english is not my native language...

OllisGit, 2020-04-13:
Hi @Protoncek ,

a lot of questions... I can only answer that are related to DLP, because I don't use "OctoDash"

- Upload: Adding Layer-Indicators is only done during FileUpload (via Browser Drag&Drop, or API called from Slicer
- Watched-Folder: I never used this mechanism. But I tested it right now. It if works as expected: copy files to "watched-folder" after a couple of seconds it is transfered to "upload-folder". 
It is working with DLP enabled or disabled. 
Start your printer in SAFE-Mode, so that all Plugins will be deactivated and then try again. 
If it still not working, raise a ticket in OP-GitHub or ask a question in forum: https://community.octoprint.org/
- Layer-Indicator: Please add your gcode (directly created by your slicer) with the added M177 INDICATORs to this ticket, so that I am able to reproduce the error.
If the slicer adds the indicators correctly no error popup should appear.

OllisGit, 2020-08-05:
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works

closed 2020-08-05 by OllisGit

Protoncek, 2020-08-05:
Damn.. sorry, it seems that i totally forgot about this topic...feedback... well, i use only update from slicer, so i have no problems since. I practically don’t use samba drive anymore, just occasionally for checking...