OllisGit/OctoPrint-DisplayLayerProgress #139: OctoPod & Print Time

Neiljclements, 2020-04-16:
Just a quick question about the OctoPod compatibility. The latest version broke it and people had to rewind to the previous version, so I was wondering when the latest version will be released please?

Secondly, there was a possibility of calculating layer time from the layer progress average with your add on. Did anything ever come of this? It has got to be way more accurate than OctoPrint calculation, which gives ridiculously stupid print times.

Thanks for a great app. Super useful 😊👍🏽

OllisGit, 2020-04-18:
Hi @Neiljclements ,

regarding OctoPod, new DLP Version 1.20.0 is out, see also https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/136

I am not sure what you mean with "there was a possibility of..". 
The current implementation doesn't support this and it also not planed to support this.
I think it is also not really exact. Just print a cone. Each top layer will be faster then the layers below.
Maybe this is already implemented in the Dashboard-Plugin from StefanCohen...I don't know.

Didi you already try this: https://plugins.octoprint.org/plugins/PrintTimeGenius/

StefanCohen, 2020-05-13:
So far the Dashboard just tracks the layer progress per layer as reported by OctoPrint.

OllisGit, 2020-05-20:
PTG integration in DLP is planed, see  https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/144

closed 2020-05-20 by OllisGit