OllisGit/OctoPrint-DisplayLayerProgress #141: Not an issue but a request for clarification

towlerg, 2020-04-20:
Sorry if this is the wrong place, I just spent a couple of hours trying to get to grips with Git and I think its addled my brain.

While waiting for my printer I've been familiarizing myself with PrusaSlicer, OctoPrint and it's plugins using a virtual printer.
I understand from your notes that I need to specify GCode to PrusaSlicer to add a Layer comment for each layer. As I tried your plugin before I read the notes I find that even without these changes I'm surprised to see that your plugin works super.

PrusaSlicer is all defaults for a Mini, your plugin is also all defaults.

;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]

;AFTER_LAYER_CHANGE
;[layer_z]

I see the G92 E0.0 in the GCode but only one occurrence of the text Layer, as in ; first layer extrusion width = 0.42mm.

I'm obviously missing something, any chance you could give a sucker an even break, and explain my misunderstanding?

[Raspberry_Pi_B+_Sleeve_v1-1_CAP_0.15mm_PLA_MINI_4h51m.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4504334/Raspberry_Pi_B%2B_Sleeve_v1-1_CAP_0.15mm_PLA_MINI_4h51m.zip)

OllisGit, 2020-04-24:
Just add ```;LAYER:[layer_num]``` into your Slicer. Then everything should work out of the box.
[image]

towlerg, 2020-04-24:
Thanks for the reply.
I guess I wasn't clear. My point was the it works WITHOUT the layer comment and I don't understand why. If you look in the attached file you'll see the absence of the text ";Layer".

Given all the back and forth your having with this plugin, I happy to let this just be one of thos inexplicable things.

OllisGit, 2020-04-25:
Hi @towlerg, 
the reason is the default-setting for Slic3r (maybe you need to scroll down in the text-area).
The plugin just counts the BEFORE_LAYER_CHANGE-comments...and It looks like that this comment is added by PrusaSlicer 2.2 as well.

```
count	[;BEFORE_LAYER_CHANGE]		Slic3r
```

closed 2020-04-25 by OllisGit