OllisGit/OctoPrint-DisplayLayerProgress #142: Layer indicator not found in file

Spacekoaster, 2020-04-22:
Hello, 

I've been using Prusa slicer 2.2.0 for a bit now, and everytime I print something, I get the following error message: 
_Layer indicator not found in file: 'filename.gcode'
Check layer pattern in DisplayLayerProgress-Settings and reupload the file!_

I don't what to do, as it was working fine when I was using Simplify 3D.

I'm using Prusa slicer 2.2.0, DisplayLayerProgress 1.20.0 and Octoprint 1.4.0

OllisGit, 2020-04-23:
Hi @Spacekoaster,

if you don't want to display layer-informations, then just disable the plugin.

Spacekoaster, 2020-04-23:
But I want to display layer information, it’s just that it doesn’t work and that I get this error message every time.

OllisGit, 2020-04-24:
How does your gcode look like?
In the gcode there must be some comments (started with ;) like this one:
- ```;LAYER:123```

Please attach the problematic gcode and also (if changed) the layer-expression from the plugin-settings

Spacekoaster, 2020-04-24:
Here's a gcode example and i haven't changed any settings in the plugin.
[covid19_headband_rc3.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4527905/covid19_headband_rc3.zip)

OllisGit, 2020-04-24:
Just add ```;LAYER:[layer_num]```  into your Slicer. Then everything should work out of the box.
[image]

closed 2020-05-06 by Spacekoaster

SM-26, 2020-09-20:
I can confirm that adding ;LAYER:[layer_num] in the custom g-code section cleared the warning when I reslice & upload a new gcode.

Sophist-UK, 2021-01-10:
Shouldn't this be in the "After layer change G-code"?

mastercko, 2021-02-06:
> Shouldn't this be in the "After layer change G-code"?

No, I think the idea is to display what layer the printer is _currently_ on, which implies that this code should happen on the "before" stage, not after.

Sophist-UK, 2021-02-06:
> > Shouldn't this be in the "After layer change G-code"?
> 
> No, I think the idea is to display what layer the printer is _currently_ on, which implies that this code should happen on the **before** stage, not **after**.

Which is exactly why I believe it should be run AFTER the layer change when the layer number has **just been incremented** and not just before the layer change when the layer number is for the layer that has **just finished**