OllisGit/OctoPrint-DisplayLayerProgress #143: Uninstalling shows Not connected

ProtomakerSprint, 2020-04-22:
Uninstalling shows something like not connected can not display layers

Using Installed Plugins in OctoPrint Settings uninstalled "Clean" method but still displays not connected can not display layers as if to suggest it is still installed but not working.

ProtomakerSprint, 2020-04-22:
Displays:
Can't get stats from DisplayLayerProgress. This plugin is required and provides GCode parsing for Fan Speed, Layer/Height info, Layer Durations and Average layer time. Is it installed, enabled and on the latest version?

OllisGit, 2020-04-23:
Hi @ProtomakerSprint ,

I am not sure what problem do you have, with the DisplayLayerProgress-Plugin.

If you uninstall the plugin other plugins like "dashboard" or "OctoDash" could not display e.g. layer-information.

Maybe you attach an screenshot to make it a little bit clear!

DefiantBidet, 2020-04-24:
this is what is shown after uninstalling:
[image]

StefanCohen, 2020-05-14:
Hi. I'm the Dashboard developer. That message is coming from The Dashboard plugin, not DLP. It is displayed since you have Dashboard installed without DLP which is required for a substantial part of the Dashboard widgets.

OllisGit, 2020-05-20:
Hi @ProtomakerSprint , @DefiantBidet ,
since Version 1.12.0 of Dashboard-Plugin, you can disable this warning:
https://github.com/StefanCohen/OctoPrint-Dashboard/releases/tag/1.12.0

closed 2020-05-20 by OllisGit