OllisGit/OctoPrint-DisplayLayerProgress #145: Problem with layer information

jjaviercs, 2020-05-01:
I saw more people have this issue, I have the last version runing:

DisplayLayerProgress Plugin (1.20.0)
OctoPrint 1.4.0

I see corretly the information ";LAYER:x" about layer information in te attached gcode script named "Sample gcode  with messange no layer DisplayLayer.txt" generated by cura.

[Sample gcode  with messange no layer DisplayLayer.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4562925/Sample.gcode.with.messange.no.layer.DisplayLayer.txt)

Any files, the exception the plugin not show this problem, The gcode files are generated with Cura or Astroprint, with astroprint it almost always happens and like the previous example it does carry the comment with the information of the layer. With cure it happens sometimes, lately most.

How can I solve the problem?

Regards

OllisGit, 2020-05-20:
Hii @jjaviercs ,

I couldn't reproduce the issue. I uploaded your file via Drag&Drop and the ```M117 INDICATOR-Layer``` is added during upload.

- How do you upload your gcode files to OctoPrint?
- Do you modify the layer-expression in the plugin-settings or do you use the default values?

Klammerson, 2020-06-23:
If I may use this thread to post the same issue with my settings!?

[CFFFP_string_test.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4818760/CFFFP_string_test.txt)

I'm using Cura 4.6.1 and have this code `;\s*LAYER:\s*([0-9]+).*` inserted using Post Processing Script before layer change.

I upload my files directly to my OctoPi via SMB.

Nothing changes, message always appears.

OllisGit, 2020-06-30:
Hi @Klammerson,
if your not using the WEB-UI or the OP-API, it did not work out of the box!

The plugin "enhance" the gocde during upload. It adds layer-indicators. If you copy the gcode via copy command, then you need to add the layer indicators by your own via your slicer
See here for more details
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works

Klammerson, 2020-06-30:
@OllisGit thx for the reply.
I did add the code within my slicer as required but still the „error“ message appears.

I have added the GCode file in the other Post.

OllisGit, 2020-06-30:
Yes, because the CURA-Postprocessor is not able to add layer-numbers to the gcode like PrusaSlicer.
[image]

It would be nice if the Postprocessor could be add such informations to the gcode:
```
M117 INDICATOR-Layer14
...
M117 INDICATOR-Layer15
...
M117 INDICATOR-Layer16
```
You can raise a request to the CURA issue tracker or ask in the OP forum for an solution..or an other Cura-Extension.

OllisGit, 2020-06-30:
Hi @Klammerson ,

I wrote a post processing script for Cura, see wiki for installation/usage: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works#cura
With this approach you can use Cura with your SMB.

Please test and give feedback!

closed 2020-08-15 by stale[bot]