OllisGit/OctoPrint-DisplayLayerProgress #146: Uninstalled, LCD still shows INDICATOR-Layer0

Cheule, 2020-05-18:
I tried out DisplayLayerProgress 1.20.  I was warned by octoprint that the plugin might cause layer shift for some users.

I uninstalled Dashboard and DisplayLayerProgress using "Uninstall and Remove Data."  I restarted OctoPrint.  However, on my next print, the LCD on my Ender 3 said "INDICATOR-Layer0" where the progress % should be.  I looked in the plugin manager for OctoPrint, and saw that under Plugin Cleanup two plugins still reported data left, ABL (Advanced Bed Level, not your plugin) and DisplayLayerProgress.  I told the plugin manager to remove both data, and it restarted.  The data was still there.  The LCD still displays "INDICATOR-Layer0."  I finally thought to reintstall your plugin, and then remove again.  Now I don't see it in the cleanup area by my LCD still says the same message.
[image]

credomane, 2020-05-18:
DLP modifies gcode files when they are uploaded by adding placeholders. These placeholders are replaced with actual values when you print the file.

Removing DLP just causes those placeholder values to get displayed. If you want those placeholders gone you will have to upload a copy of your gcode from before it was sent to octopi. With DLP disabled as well of course.

Cheule, 2020-05-18:
Oh, I had no idea!  That makes a lot of sense.  I'll try resending a g-code file and see if the message goes away on the LCD and report back.  In the middle of a long print, so it will be some hours.

OllisGit, 2020-05-20:
Hi @Cheule ,

more technical informations could be found here: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works

@credomane thx, for answering!

closed 2020-05-20 by OllisGit