OllisGit/OctoPrint-DisplayLayerProgress #147: [Enhancement] Log layer times by name

DonaldLSayers, 2020-05-21:
I am tweaking print time efficiency in my slicer, and I'm trying to log not only print times but time by layer to compare where i am gaining or losing time.  

It would be cool to have the layer times logged in a sheet . and maybe color code them based on best/tied times

example

**File Name**

layers | Date1 | Date2
--- | --- | ---
1 | 10s | 9s
2 | 10s | 9s
3 | 15s | 10s
4 | 10s | 9s
5 | 12s | 10s
6 | 10s | 9s
7 | 10s | 9s
8 | 10s | 9s

Thanks for your work on this! it is great.

OllisGit, 2020-05-21:
Hi @DonaldLSayers ,

well.....if you are able to implement your own plugin, it is "easy peasy" ;-)

Just listen on the event "DisplayLayerProgress_layerChanged" and grab the layer-number, the duration and write everything into a log file.....If you want to start with plugin-development I can coach you ;-)

The "Layer Duration Graph" from the [Dashboard-Plugin](https://github.com/StefanCohen/OctoPrint-Dashboard) is not exact for you?
[image]
...maybe you can (also) raise your request in the Dashboard issue-tracker.

Because I am not sure who else need this information and is it worth to implement it.

OllisGit, 2020-08-05:
Note to myself: Could be a feature for "PrintJobHistory"- or "InfluxDB"-Plugin

closed 2020-09-14 by stale[bot]