OllisGit/OctoPrint-DisplayLayerProgress #148: No more "Layer Durations"

energiebrand, 2020-05-21:
I am running Octoprint 1.4.0 on Octopi 0.17.0.
Display Layer Progress is 1.21.0

With this configuration "Layer Durations" Graph is non more shown. Same behaviour on another RPi with Octoprint 1.4.0 on Octoprint 0.16.0
Sometimes an empty graph is shown, sometimes nothing

[image]

[image]

OllisGit, 2020-05-21:
Hi @energiebrand ,
I debugged the Dashboard plugin and all DLP Layer-Values were sent to the Frontend.
I created a bug, so that StefanCohen  could take a look in it: https://github.com/StefanCohen/OctoPrint-Dashboard/issues/160

energiebrand, 2020-05-23:
Thanks a lot! That means the problem is sitting in the "Octoprint Dashboard"? If so, sorry for opening a ticket here...

StefanCohen, 2020-05-25:
Thanks. I received the ticket from Olli and will try to reproduce on my own.

OllisGit, 2020-05-25:
Hi @energiebrand ,

please attach the octoprint.log file to this issue, because I could identify the issue or at least the same behaviour (seeDashboard-Ticket for more details).

Maybe there is a similar error in your log-file.

energiebrand, 2020-05-25:
This is the file from today. If you need more from the last days please let me know.

[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4678436/octoprint.log)

OllisGit, 2020-05-25:
hmmm...I found a couple of errors in your log, but nothing is related to DashBoard/DLP-Plugin.

So, we need to debug your UI. Are you familiar with JavaScript-Debugging?
..if not, I will teach you.
Please do the following:
1. Create a textfile "LayerDuration.gcode" with the following content:
```
M117 INDICATOR-Layer1
G4 S5
M117 INDICATOR-Layer2
G4 S10
M117 INDICATOR-Layer3
G4 S15
M117 INDICATOR-Layer4
G4 S10
M117 INDICATOR-Layer5
G4 S5
M117 INDICATOR-Layer6
G4 S3
M117 INDICATOR-Layer7
M117 finished
```
1. Open your OP-WebSite (use Chrome, if possible) and upload the gcode via drag and drop
1. Only select this file for printing
1. Press F12 to open the Chrome-Developertools
1. Click on the Sources-Tab in the top
1. Select the "packed_plugins.js" file (see left-tree)
1. Ctrl + f: In the search field enter ```if (data.layerTimes```
1. Create a break-point by pressing the left mouse-button left to the line number (red dot should appear)
1. Start print
1. A couple of times the browser stops and you will see a "play" symbol on top
1. Close the "Call Stack" and open the "Scope"-Ribbon on the right, to inspect the data which was sent from the server
1. The "data" must include "layerTimes" and "layerLabels" to print the duration graph

An other test:
- Also leave the Developer-Tools open and press Ctrl + r for reloading the page. In the Console-Tab (bottom) there should no red error message appear.

energiebrand, 2020-05-26:
OK, thanks I will do soon. Currently I´m printing...

energiebrand, 2020-05-26:
OK, tried on another printer. Came until point 7 of your description (screenshot) but now in the Crome browser I get a PopUp "Paused in debugger" and cannot start or doing anything. When pressing the blue arrow (resume) something in the debugger windows happens but after that same situation. cannot control the printer via GUI. I think I am doing something wrong here...

When opening another browser (Firefox) everything is controlable.
[image]

energiebrand, 2020-05-26:
...in addition to the little help I need (have a look at my previous post) I would like to inform you, that a downgrade of "Display Layer Progress" plugin to 1.20.0 and 1.19.1 does not change anything. SoI upgraded finally again to 1.21.0.  ...and still would like to perform your proposal above, but stuck at point 7, not able to start printing without a little help...

OllisGit, 2020-05-26:
Hi @energiebrand ,
in my documentation you need to switch 8 with 7, sorry....anyway:
Just remove the breakpoint (click on the red dot), start print and after the print is started/running just add the breakpoint immediately.

Now "each second"  the browser stops. Please look into the data-section for layer Informations  (in your screenshot the "layerLabels" are an empty array ```[]```.
Just press "play" to continue.
At the beginning of the print everything is empty, but in the middle of the print the layer-values should be changed.

One assumption from me and Stefan was, that a sub-module called "psutil" is the root case, but I can already see in your screenshot that the  cpuPercentage could be evaluated by the psutil...hmmm..strange.

OllisGit, 2020-05-26:
In your last log I could not find any issue related to DLP or Dashboard.
There is only an Issue with the Status-Line Plugin.

Your gcode include this command:
```
M117 PLA wird vorgeheizt auf 45/200°C
```
But the plugin struggle with the "celsius-circle", you should remove this symbol.

energiebrand, 2020-05-26:
Thanks a lot for fast feedback! 
How should I say... I have continued my investigations in parallel and now I am pretty sure that everything has to do with the PlugIn "Custom Control Editor". I use this to create some buttons like "PreHeating" and (funny...) one button includes also the command "M117 PLA wird vorgeheizt auf 45/200°C" as a message followed by M190 S40 & M109 S205 . There are also some other buttons with shorter commands.
When I remove this PlugIn everything works fine. I did open an issue some time before when the owner released a new version. I installed this and all buttons were gone. I did not inspect the YAML. I created new buttons, saved them and they ware working. After I made a reboot all buttons were gone again. So I created an issue (#32) about the new release. There are some other guys who had the same or similar problem. See issue #31 as well as some other guys followed me in issue #32 having the same problem.
Then I went back to the prevoius release and eveything was fine...but now with problems with the Dashboard and the new releases.
I am not a SW developer and cannot describe what happens here and even I cannot say if the Custom Control Editor does really disturb the Dashboard. However when I remove it the Dashboard works 100% so far I can see. With the latest Custom Control Editor now installed it does not. 
Now in the last step I will fallback again to the old release of the CCE to see if it works.
Maybe you will install it on your RPi and check in parallel if this PlugIn has an imapct?

energiebrand, 2020-05-26:
In addition...I stopped for today. Last status of one of my RPi´s is, that everything is working now! All PlugIn`s including the latest Custom Control Editor works as expected. Also the problems I had with the latest release of this one seems to be gone. Interesting that one guy added the same problem I had with this PlugIn yesterday again.
Tomorrow I will do the same with my other RPi and give a feedback. So maybe the Bugs can be closed, but please give me time until I can confirm this. Everything is too strange...

energiebrand, 2020-05-27:
Octoprint 1.4.0 on Octopi 0.17.0 and Octoprint 1.4.0 on Octopi 0.16.0
I can confirm that issue #148 can be closed from my point of view.
Also from my point of view issue #163 #164 of the Dasboard can be closed (I will give a comment there). I propose to close issue #160 in Dashboard.

Octoprint 1.4.0 on Octopi 0.17.0
What did I?
Removed Dashboard, Display Layer Progress and Custom Control Editor including data.
Sudo apt-get update & upgrade (just routine...)
Installed Dashboard, Display Layer Progress and Custom Control Editor (all latest revisions)
Defined Custom control buttons in Custom Control Editor, saved, shut off, boot
Now everything works, all problems are gone!

Octoprint 1.4.0 on Octopi 0.16.0
Did the same as above but -> the issue #31 & 32 of the Custom Control Editor are still present
I did a fallback to the old release of Custom Control Editor; here it works!
However no more problems in combination with Dashboard and Display Layer Progress!

OllisGit, 2020-05-27:
Allright. Closing this issue.

closed 2020-05-27 by OllisGit