OllisGit/OctoPrint-DisplayLayerProgress #149: No layer information with latest release...

jbuszkie, 2020-05-21:
I just upgraded to the latest, I believe..  Now when I try to print something new, It complains it can't find any layer information.  If I print an older design that was already in Octoprint, It doesn't complain.
I don't believe I changed anything in my slicer (Slic3r)
I see the layer info in the GCODE...
`G92 E0 ; reset extrusion distance
; Filament gcode
;BEFORE_LAYER_CHANGE
;G92 E0.0
;0.2

G1 Z0.200 F9000.000 ; move to next layer (0)
;AFTER_LAYER_CHANGE
;0.2
G1 E-2.00000 F3000.00000 ; retract
`

OllisGit, 2020-05-21:
Hi @jbuszkie,

please check your DLP-Plugin Settings, maybe the "Add Layer-Indicators during upload" is not checked.

[image]

jbuszkie, 2020-05-21:
Did the update uncheck this by default or was this new from this version? It's not checked.  I'm in the middle of a print but I'll "check" this out after!  Thanks

jbuszkie, 2020-05-22:
I don't know how this became unchecked..  But after checking it, it now works as before.

closed 2020-05-22 by jbuszkie