OllisGit/OctoPrint-DisplayLayerProgress #152: Wrong Total Height

CyberExplorer, 2020-05-30:
If I download calibration cube from thingverse and slice it cura its shows correct 20mm height. But even simplest parts from solidworks and cura then displaylayerprogress and all other plugins using value from DLP is wrong. By the way cura shows correct height.

I send a sample.

[Test.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4705197/Test.zip)

CyberExplorer, 2020-05-30:
[image]

CyberExplorer, 2020-05-30:
Its almost finished part. Only 2 layer left but still shows wrong height. Actual part height is 5mm.

OllisGit, 2020-05-30:
Hi @CyberExplorer,
I looked into your GCode and the Object is only ~5mm height.
I used http://gcode.ws/
[image]
The model info in gcode.ws shows only 15mm, instead of 20mm.

I imported the gcode to Cuar4.5.0 and I can only see a 5mm object.
[image]

If you look into the gcode via texteditor, you find the last layer in at ```Z4.9``` in line 15340. 
DLP shows ```5.5```, because after the last layer there is an other relative z-movement +0.5.

This is wrong, should be ```5.4``` no idea why.

But, please verify your model. Maybe you printed/downloaded the wrong Cube.

CyberExplorer, 2020-05-30:
You misunderstand me. If I download a calibration cube from thingverse everything is fine. 20x20x20 cube shows 20mm on cura and your plugin. There is no problem with that.

But if I export my designs from solidworks as STL. And slice it Cura then the problem begin. Cura show correct height. Like in screenshot its show 5mm. But DLP show it 15mm. And Job Progress is wrong to because of layer height.

[image]

OllisGit, 2020-06-12:
ahh..okay!
Which method do you use for evaluating the height?

[image]

Maybe you need to switch to the second option to read the correct value from the gcode. Because there is a movement higher then the object itself.

OllisGit, 2020-08-05:
no feedback. assumption: it works now.

closed 2020-08-05 by OllisGit