OllisGit/OctoPrint-DisplayLayerProgress #153: Fixed total height at 123 layers / 20mm

markus1802, 2020-05-30:
Hi,
first the total height wasn't displayed at all. I then reset the setttings in the octoprint plugin settings dialog. Since then the total layer height is always displayed as 123 and the total height as 20mm. This is even if the model is way larger. See attached screenshot.
Any idea what may be wrong here?
Thanx
Markus
[image]

markus1802, 2020-05-30:
This is a snippet of the gcode from the above screenshot. I downloaded it from octoprint after printing.
...
M83 ;relative extrusion mode
G1 F1500 E-2
;LAYER_COUNT:293
;LAYER:0
M117 INDICATOR-Layer0
M106 S51
...

markus1802, 2020-06-04:
New variant: the total layers are always displayed as 45. Total height still is 20.0 on every print. 

The print in the screenshot has 490 total layers.

[image]

OllisGit, 2020-06-04:
Very strange!
Can you please attach the original and the modified gcode downloaded from OP to this issue?

Then I can debug the processing during print.

markus1802, 2020-06-04:
Thank you for your answer! 
I understand that the file is changed during print (from OctoPrint?). I use "Print with OctoPrint" from Cura. Therfore I usually don't have the file before printing. Because of that I attached a zip with the file downloaded from OctoPrint (while the print currently is still printing, it's a larger print) and a file I just now saved to file from Cura (the object with unchanged settings was still opened). Hopefully this helps. Otherwise I have to make a new test when this print finished.

markus1802, 2020-06-04:
I just compared both files by myself. I used german characters in the filenam. Stupid! But I think that's not the problem because I had this behaviour with other files also. The only difference I found is an added:
M117 INDICATOR-Layer0
on every layer.

;LAYER_COUNT:490 also looks good for me.

OllisGit, 2020-06-04:
hmm..I just uploaded the file ```_from_cura``` and everything looks fine after file selection.
[image]
Also the modified file you provided includes 490 layer-indicators.

After uploading to my test-printer, the ```;MESH``` comment is not modified. This is not a "feature" of my plugin. I have no idea if the printer can handle this command: ```@Object D??bel-Hilfe1.stl``` 

What other plugins do you use?

You already executed the reset-settings button, but please take a look directly into the OP ```config.yml``` file.
The location could be found in the Folders-Section. Use the Upload Folder and go one directory above. In that root folder there should be the config.yml file.
If the reset was executed successful, you should not find any entry for DisplayLayerProgress
```
plugins:
    DisplayLayerProgress:
        layerOffset: '1'
```

markus1802, 2020-06-05:
I did some more testing:

1. Yesterday I got an update to the Cura-OctoPrint-Connection. Because of that I first made a new test print -> 45 total layers

2. config.yaml showed:
    DisplayLayerProgress:
        layerOffset: '1'
        showAllPrinterMessages: false
So I made a new Reset Settings. Test print -> 45 total layers

3. I searched for ;MESH and @Object: must be a "feature" of CancelObject-Plugin. So I uninstalled CancelObject and restarted OctoPrint. Test print -> 489 total layers

4. Installed CancelObject again. Test print -> 489 total layers

Ok, that's really strange. Obviously CancelObject was the cause. Anyways, it's working again. Thank you for your support and sorry for bothering you.

markus1802, 2020-06-07:
Hi,
unfortunately I have to reopen this issue. As of my comment above the last print data looked correct. But the total layer got stuck at this 489 total layers for all following prints.

I now have totally uninstalled the CancelObject-Plugin. The DisplayLayerProgress-Plugin I have uninstalled (with deleting data) and reinstalled. 

Result ist:
[image]

The printed object has a total layer count of 42 and a total height of 8.4mm. From the beginning of the print till it's end OctoPrint shows 20mm and 52 layers. I lastly didn't print anything with total layers of 52. Totally confusing.

I add the gcode files:
[gcode2.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4742602/gcode2.zip)

Do you have any idea what I may check?

markus1802, 2020-06-07:
Ok, I uninstalled more Plugins:
- Floating Navbar
- Printoid

Now, the total layers are displayed correct. I started several test prints with larger oder smaller objects and the total layers were always as expected. But the total height still shows 20.0mm if the object is smaller than 20.0mm. If it's larger the expected size is shown.

OllisGit, 2020-06-12:
Hi @markus1802 ,
I will install the mentioned. plugin and try to reproduce the issue.

JayceeB1, 2020-06-20:
Hi

also got some strange issues using Cura 4.6.1

[image]

OllisGit, 2020-06-30:
Hi @JayceeB1 ,

please provide the gcode downloaded from OP, maybe it is a different issue.
Do you also use the Plugin "Cancel Object"?

@markus1802 I installed "Cancel Object" and now I can reproduce the error. The max layer count is not changed anymore.
I will start analysing the issue a little bit deeper the next days.

OllisGit, 2020-08-05:
fyi: I only could reproduced the behaviour once. Now everything is working. Hard to find the issue.

closed 2020-11-16 by stale[bot]