OllisGit/OctoPrint-DisplayLayerProgress #154: Slow uploads with plugin enabled

EddieGreen, 2020-06-22:
As per thread in [Octoprint](https://github.com/OctoPrint/OctoPrint/issues/3437#issue-557683018) I resolved this by disabling this plugin.

What seems to be happening is that the upload completes and DLP starts working on the new file. In the meantime OctoPrint's display continues to read 'saving...' which may mislead one into thinking that there is a problem with OctoPrint's upload speed.

It's apparent that DLP is doing a lot of work at this point. A side issue also occurs when uploading while a print is in progress, which brings printing to a virtual standstill while it works.

I don't think any plugin should hijack the host with such a CPU intensive task, and it should defer processing if printing is in progress. Adding a message and progress indicator would clearly identify where the hold-up comes from.

Can these suggestions be implemented?

msteele999, 2020-06-23:
I am having this issue as well - however it started AFTER I created a new OctoPrint install on Windows - I went from Python 2.7 to Python 3.8.3 - and that's when the issue started. Here is a snippet during upload - the file took about five minutes to complete the upload and my system was unresponsive during that time.

Turning OFF 'Add Layer-Indicators during upload ' resolves the issue.

`2020-06-23 16:40:09,240 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. Checking LayerExpressions.
2020-06-23 16:40:09,240 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. LayerExpression valid. Start processing...
2020-06-23 16:41:28,137 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-06-23 16:46:49,908 - octoprint.filemanager.analysis - INFO - Starting analysis of local:ASP_Elite_Dangerous_Cut-mod_.12R.gcode
2020-06-23 16:46:49,908 - octoprint.filemanager.analysis - INFO - Invoking analysis command: c:\octoprint\venv\scripts\python.exe -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 C:\Users\mstee\AppData\Roaming\OctoPrint\uploads\ASP_Elite_Dangerous_Cut-mod_.12R.gcode
`

Available for testing so please let me know.

OctoPrint 1.4.0
DisplayLayerProgress 1.22.1
Python 3.8.3
Windows 10 32bit
Intel Compute Stick
Sovol SV01 via USB (COM3)

OllisGit, 2020-06-30:
Hi @EddieGreen ,

thanks a lot for linking the issue from OP. Also mentioned in [OctoPrint on Air](https://www.youtube.com/watch?v=E6i_X_63s50&t=20m42s)) this fix will be available in OP 1.4.1.

In the meantime, as @msteele999 mentioned, turn off the "Add Layer-Indicators during upload" and put the ```M117 INDICATOR-Layer``` via your sclicer to your gcode.
More details could be found here: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works

msteele999, 2020-06-30:
Thanks for the update.

Regarding adding the M117 code to Cura for example, do you know if this is
a post-processing addition (like doing search and replace to add the code)?
Or is this a 'start g-code' entry?

I'm a tad confused about how to facilitate this.

msteele999, 2020-06-30:
Disregard - I found it here:
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works

msteele999, 2020-06-30:
Actually - that does not work in Cura 4.6.1 - investigating

msteele999, 2020-06-30:
This works:

Just use the build in "Search and Replace" Script

   - Search: ;Layer:
   - Replace: M117 INDICATOR-Layer
   - Leave reg ex unchecked

Klammerson, 2020-07-01:
Actually the script needs to be placed in cura/Resources/plugins/plugins/PostProcessingPlugin/scripts

Now everything seems to work like it is supposed to. nice work! thanks a lot!

closed 2020-08-05 by OllisGit