OllisGit/OctoPrint-DisplayLayerProgress #155: Issues with using plugin with DaVinci 1.0A with .92 of Repetier

mpoffo1231, 2020-06-24:
I slice primarily in Simplify 3D and Cura.  I am having issues with getting this to work with files sliced in Simplify 3D on a modified DaVinci 1.0A w Repetier .92.    I have 2 identical Octoprint setups running.  Both run on a Pi 4B with the latest octoprint and same plugins.  I have been using one on my Ender 5 Plus with no issues.  The plug-in works as advertised.

However for whatever some reason with the Octoprint set-up (same as Ender 5 Plus setup) on my Davinici it will not display the layer if I slice with Simplify3D.  I get the following error: "Layer indicator not found in the file: xxxxxx.gcode Check layer pattern in DisplayLayerPorgress - Settings and reupload the file!".  I look at the layer tab and there really is nothing to change other than check or unchecking some boxes.  

Start and end gcode are the same with Simplify 3D and Cura.  I am missing something that I need to edit somewhere in the slicer or the plugin?  If I upload a file sliced with my Ender 5 Plus profile for Simplify 3D to the DaVinci it is fine (though not the right printer).  No errors.  Files sliced in Cura work with the same start/end gcode/scripts.

Any advice would be appreciated.

mpoffo1231, 2020-06-24:
So I decided to do a new printer set-up/process in Simplify3D using "other" printer set-up rather than the DaVinci 1.0 option.  It appears that took care of the issue.  The process I had been using from that printer goes back a ways.  In retrospect the DaVinci 1.0 option in Simplify 3D is for the stock printer with stock firmware.  There must have been some setting that caused the plug-in to trip up using this stock option from Simplify 3D.

closed 2020-06-24 by mpoffo1231

OllisGit, 2020-06-30:
See https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works maybe it helps