OllisGit/OctoPrint-DisplayLayerProgress #156: Error message "Layer indicator not found in file! "

dabossie, 2020-06-27:
I'm using Prusaslicer 2.2.0 and get this message in my octoprint plugin 
[image]
Where can I solve this problem in Prusaslicer?

dabossie, 2020-06-28:
**Think i've found it.**

PrusaSlicer 2.2.0 does not add a "layer-indicator" by default. In Print Settings, Output options, you can check "Verbose G-code" which will add (along with a lot of other stuff):

**G1 Z0.350 F7800.000 ; move to next layer (0)
G1 Z0.650 F7800.000 ; move to next layer (1)**

Or you can go to Printer Settings, Custom G-code, Before layer change G-Code, and add (for example):

**;LAYER:[layer_num] Z=[layer_z]**

which should be matched by the CURA setting.

Klammerson, 2020-06-30:
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/145#issuecomment-648055338

for some reason this doesn't work for me in Cura

OllisGit, 2020-06-30:
@Klammerson see #145 
@dabossie thx, I added that info to the [wiki-page](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works)

closed 2020-06-30 by OllisGit