OllisGit/OctoPrint-DisplayLayerProgress #157: [estimated_end_time] not taking into consideration timezone or Summer Time

comm, 2020-07-01:
When displaying [estimated_end_time], it is 1 hour behind current time (DLST/Summer Time?) of the Octoprint server (RPi) when on the LCD or desktop. However, the Dashboard widget, which relies on DisplayLayerProcess, shows the estimate end time time correctly.

comm, 2020-07-01:
Apologies all, this was related to my System Timezone being set incorrectly.

I used https://www.wikihow.com/Change-the-Timezone-in-Linux to determine my local timezone, then SSH'd into the Octoprint server and set it correctly by following https://linuxacademy.com/blog/linux/changing-the-time-zone-in-linux-command-line/.

closed 2020-07-01 by comm

Ashoat, 2020-12-22:
Hi @comm – apologies for reaching out in this unusual fashion. Is there any chance I can get in touch with you?

The context is that I'm working on an open source project to make E2E encryption work beyond simple chat apps. We're starting off by trying to build something like an E2E-encrypted self-hosted Discord/Slack alternative. Long-term we actually want to make a platform where anything – from Facebook, to Dropbox, to Reddit, etc. – can be implemented in an E2E-encrypted fashion, where user data is owned by users and self-hosted instead of being hosted on corporate servers in the cloud.

I really like the name "Comm" and would be interested in buying your GitHub username. My email address is ashoat@gmail.com – please feel free to contact me there!