OllisGit/OctoPrint-DisplayLayerProgress #158: [Enhancement] Show layer information in file pane

UserBlackBox, 2020-07-05:
This is a suggestion to add layer information to the file details under the file browser pane in OctoPrint so that it is possible to view how many layers are in a gcode model

OllisGit, 2020-08-05:
Hi @UserBlackBox ,

nice idea, but I think today the filebox is really packed:
[image]

Also, it took a while evaluating the layers from each gcode, so the browser is a little bit busy...okay, I can add such information into the meta-store.

hmm...maybe also showing the height (or dimensions).....could be easy to implement...will see.

OllisGit, 2020-08-16:
Hi @UserBlackBox ,

I played around with different approaches. Maybe one possible way is to overwrite the ```getAdditionalInfo``` function.
But each time you want to see the layer-info, you need to click on the expand button. 
Maybe there is a setting to expand the section as default...but I didn't found it.

What do you think, is it worth to implement it in that way?
[image]

OllisGit, 2020-08-16:
...hmm..it is not possible to get the layer-values to the frontend. 
I will ask in the OP-Forum, maybe someone has an idea.

OllisGit, 2020-08-17:
Hi @UserBlackBox ,
I found a solution to implement it. 
After installation of the newest version 1.23.0 you can see the total layer count after adding new files or selecting an old file and refresh the view.

Please test and give a feedback.

Thx, in advance
Olli

UserBlackBox, 2020-08-18:
I have updated to 1.23.0 and it works well. Thanks

closed 2020-08-18 by OllisGit