OllisGit/OctoPrint-DisplayLayerProgress #159: Total Height always 10.0 mm

spanzetta, 2020-07-23:
Hello
I noticed that it always shows 10.0 mm when objects are lower then this height...
When they are taller... it shows the right value..
See screenshot below
Are you aware of such problem? 
I currently have the Height method set to "read from G-Code .."
Thanks in advance

[image]

OllisGit, 2020-07-23:
Hi @spanzetta ,

if possible, please also provide the affected gcode and the octoprint log file, maybe there is an error listed.

BR
Olli

spanzetta, 2020-07-23:
Thanks for the reply..
see attached files
Thanks

[log_and_gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/4967695/log_and_gcode.zip)

spanzetta, 2020-07-25:
Any news?
Tha

spanzetta, 2020-08-01:
hi Olli, 
did you have the time to look at this?
thanks in advance

spanzetta, 2020-08-05:
Olli
Are you there?
Thanks

OllisGit, 2020-08-05:
Hi @spanzetta ,

sorry for the delay, was busy with my other plugins.

What "Height-Methode" do you use in your plugin settings?

Maybe "Max Z-Value in G-Code"
Change to "Max Z-Value with Extrusion on Z-Layer" then you will see for your example gcode 3.0mm (
[image]

spanzetta, 2020-08-05:
As written in my original post I currently have the Height method set to "read from G-Code .."

I will try the other method

Thanks a lot

CmdrCody51, 2020-08-06:
> As written in my original post I currently have the Height method set to "read from G-Code .."
> 
> I will try the other method
> 
> Thanks a lot

Actually I think this highlights a different bug. At the end of you G-Code in your end code you state:

G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

So - actually the Height should be 13.2.
Using OllisGit suggestion should make it read 3
I think the real problem is that the plugin doesn't handle Incremental very well.
Not even doing a print - goto Control and click the Z axis Up and Down - No Down....

spanzetta, 2020-08-06:
sorry but the GCode (made with Cura 4.6.1) is for a 2.8mm height model..
Why it should read 13.2?

CmdrCody51, 2020-08-06:
> sorry but the GCode (made with Cura 4.6.1) is for a 2.8mm height model..
> Why it should read 13.2?

Your last Z is 3. At the end you do a incremental retract and Z0.2, do the wipe out and then raise Z 10 more.
3+0.2+10=13.2

spanzetta, 2020-08-07:
I think you are referring to raise Z (10) end-code that Cura adds to every print to move up the head after the print is finished
That should not be calculated as model Z.. it's just a head movement at the end of the print..
In fact Octoprint doesn't consider that .. if you look at my screenshot (at bottom) you see the proper model Height (Z = 2.80 mm)

CmdrCody51, 2020-08-07:
> I think you are referring to raise Z (10) end-code that Cura adds to every print to move up the head after the print is finished
> That should not be calculated as model Z.. it's just a head movement [...]

And if you look at the G-Code:

;MESH:NONMESH
G0 F300 X176.083 Y38.881 Z3  <<<<<<<<<
G0 F9000 X175.68 Y39.104
G0 X174.598 Y39.105
G0 X170.517 Y40.997
G0 X165.888 Y41.791
G0 X165.878 Y17.857
G0 X122.139 Y16.847
G0 X81.753 Y16.847
G0 X27.869 Y22.347
G0 X24.181 Y85.499
G0 X22.595 Y85.5
G0 X23.046 Y128
G0 X23.136 Y128.3
;TIME_ELAPSED:18394.373503
;LAYER:14
M117 INDICATOR-Layer14  <<<<<<<<<<<<
;TYPE:WALL-OUTER
;MESH:3 edited.stl
G1 F3000 E3602.85833  <<<<<<<<<<
G1 F1500 X22.515 Y128.3 E3602.87899

Even the Preamble says:

;FLAVOR:Marlin
;TIME:19161
;Filament used: 12.4071m
;Layer height: 0.2
;MINX:16.565
;MINY:16.847
;MINZ:0.2
;MAXX:218.786
;MAXY:218.153
;MAXZ:3   <<<<<<<<<<<<<<<,
;Generated with Cura_SteamEngine 4.6.1
M140 S55

Your last extrrusion is at Z=3 going into Layer 14.
I think if you printed this and at the end did a manual M114 the Z would be 13.2

OllisGit picture with the three choices are:
  Biggest Z (10 if all your parts are less than 10 because of the 10mm lift at the end)
  Highest Z with extrusion 3 in this case
  Read the MAXZ again 3 in this case

But the machine will end up at 13.2.

spanzetta, 2020-08-07:
The option "Max Z-Value in G-Code" should NOT take in consideration the end-code (post print movement).. don't you agree?

In any case why it should consider 10 for less then 10 (for the 10mm lift at the end) .. and the real value when more then 10 mm (not considering the 10mm lift at the end)????

CmdrCody51, 2020-08-07:
> The option "Max Z-Value in G-Code" should NOT take in consideration the end-code (post print movement).. don't you agree?
> 
> In any case why it should consider 10 for less then 10 (for the 10mm lift at the end) .. and the real value when more then 10 mm (not considering the 10mm lift at the end)????

Absolutlely Not! If you want the top of the model - use Max-Z with extrusion or the slicers MAXZ statement.
I was simply trying to explain WHY you get the "Total Height always 10.0 mm" and saying it should be 13.2!
Any Z on a model the that is say 25mm tall will swamp the Z lift of 10 at the end. Because the plugin doesn't track where the Z is in incremental.

spanzetta, 2020-08-07:
It's not me that want the top of the model..
It's the plugin (it's its purpose) that should get the (right) Max-Z from the GCode.. 
so it should try to understand it properly.. not considering the post-print 

but again.. here there is anyway a problem.. because if the plugin consider the post-print movement..  it should do it always.. and not only for the less then 10mm prints.. 

I hope I was clear enough...

OllisGit, 2020-08-14:
Hi @spanzetta!

> The option "Max Z-Value in G-Code" should NOT take in consideration the end-code (post print movement).. don't you agree?

The OctoPrint GCodeProcessor can't separate between "object-gcode" and "post-gcode" it just receive each gcode line by line.

> It's not me that want the top of the model.

But maybe some other user needs this information of the Max Z-Value in the gcode :-)

> but again.. here there is anyway a problem.. because if the plugin consider the post-print movement.. it should do it always.. and not only for the less then 10mm prints..

hmm...this should always be the case! Do you have an other example gcode and screenshot where this is not the case?

OllisGit, 2020-08-17:
Hi,
I dropped the whole "height-detection" in the newest release V1.23.0. Now the value is picked up from the OctoPrint MetaData.

Please test and give feedback.

Thx, in advance
Olli

CmdrCody51, 2020-08-17:
> Hi,
> I dropped the whole "height-detection" in the newest release V1.23.0. Now the value is picked up from the OctoPrint MetaData.
> 
> Please test and give feedback.

For my part Olli this looks to be equivalent to highest level with extrusion. With the small problem that for whatever reason starts offset at the first layer height. Theoretical example:
blah blah
G0 X20 Y20 Z0.2
G1 Y25 E2
blah blah
G1 X20 Y20 Z10
G1 Y25 E35
end of extrusion gcode
The height displayed from metadata 9.8 (just try it with a calibration cube)
Personally I liked the choice of which method to use, made it more flexible.
**Maybe just default it to that.**
The problem with largest Z in Gcode was processing the incremental/relative moves and adding/subtracting them from the absolute location. The plugin still does this on the display Current Height: still only goes up and not down when manually jogged.
I'm going to check now if I do an extrusion @ Z0 off the table if the number comes up right - then it is a OctoPrint bug.

OllisGit, 2020-08-30:
Hi @CmdrCody51,
I am a little bit confused.
If you say "came out perfect for the part", then my assumption is everything is fine and I can close the issue from my side.

If you think the metadata values should be "adjusted", feel free to raise an issue on the OP-side.

CmdrCody51, 2020-08-30:
> Hi @CmdrCody51,
> I am a little bit confused.
> If you say "came out perfect for the part", then my assumption is everything is fine and I can close the issue from my side.
> 
> If you think the metadata values should be "adjusted", feel free to raise an issue on the OP-side.

Sorry to confuse - I do that a lot... So - I do not know what metadata is available from OP and the purposes it is used for, but from the way you report it in the plugin it is returning the max Z with extrusion minus the first layer height. A calibration cube 20x20x20 will report 19.7 if the first layer is 0.3mm. The Z height with extrusion would report 20.0 correct? And if you still had the 3 options as before - the largest Z would report 20.0 correct?

IF OP metadata contains First layer height - just add that to the current metadata and you should be perfect for the original poster's (spanzetta) complaint.

However - I noticed that in manual from the control tab that the plugin only adds Z moves and does not track the Z position.
I know that you are 'looking for max Z' and see some sense in that. But I and I believe many use the display to find out what the printer is doing. And now if I zero the printer and incrementally move up 10mm manually and then incrementally down 5mm the display stays at 10mm. If move incrementally move up 10mm - the display goes to 20mm while the printer reports 15mm.

This is really a topic for a different PR but it is all tied together with the original poster's question. "Total Height always 10mm" because he used the default Max Z option and the largest Z was the 10mm incremental move at the end. Problem being it was an incremental move of 10 from Z height with extrusion AND the number should have been 10mm more than Z Height with extrusion. Same problem of not tracking the machine position. I know it would slow down pre-processing but it's just the right thing to do.

Anymore you need - let me know.

CmdrCody51, 2020-09-01:
> Damn it - you made me look... You are using the metadata under "dimensions" - try using "printingArea".
> "maxZ" to be precise.
> 
> But this will not fix the incremental/absolute tracking but that isn't this PR.

I think you missed my edit....
Sorry.

spanzetta, 2020-09-05:
Hi
I can confirm that now the problem doesn't happen anymore..
I printed a 7.4mm height part and it wasn't displaying 10mm like in the past 
Thanks a lot

closed 2020-09-05 by spanzetta