OllisGit/OctoPrint-DisplayLayerProgress #162: Day of week in [estimated_end_time]

mikulik86, 2020-08-07:
I propose that if [estimated_end_time] is more than 24 hours, it should be displayed with day of week. For example: "Friday 18:36".
Better yet, "Tomorrow 18:36" if the ETA is the next day.

OllisGit, 2020-08-07:
Hi @mikulik86 ,

because everybody wants a different output, I implemented the option to format the output by your self.
In your case you just need to change the time format to ```%A %H:%M'.
You cant test it here: https://trinket.io/python/b2055e0d6d
[image]
The explanation of the format could be found here: https://www.programiz.com/python-programming/datetime/strftime#format-code

Btw. "Tomorrow" is not possible.

mikulik86, 2020-08-07:
Thank you! 
Could you just point me on how to use it? Because i have no idea :D

OllisGit, 2020-08-14:
Hi @mikulik86 ,
just go to the Plugin-Settings
[image]
Result:
[image]

mikulik86, 2020-08-17:
Totally overlooked it, sorry. Should have go through the settings more thoroughly. Nice implementation though 👍
Thanks!

closed 2020-08-17 by mikulik86