OllisGit/OctoPrint-DisplayLayerProgress #163: wrong percentage??

spanzetta, 2020-08-09:
Which one is wrong?
;-)

Is it 96% or 99%?

[image]

spanzetta, 2020-08-10:
even worst here..

what's happening???

[image]

allenrowand, 2020-08-11:
Same issue here. I used Themeify to hide the progress bar, then added the DLP progress percentage next to the layer count in the State sidebar so that it matches the printer's display. Not perfect, but at least I see a consistent percentage now.

OllisGit, 2020-08-14:
Hi @spanzetta , @allenrowand ,
which OctoPrint Version and Python Version do you use?

Please also provide the octoprint.log, maybe there is an error occurred.

allenrowand, 2020-08-14:
I'm on OctoPrint 1.4.2, Python 2.7.16, OctoPi 0.17.0.

I've attached my logs from yesterday, there were several prints and attempted prints. Thanks!

[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5077279/octoprint.log)

The-EG, 2020-08-14:
I'm seeing the exact same issue on a brand new install of OctoPrint, basically the same setup as allenrowand (I came here to post an issue and found this!)

OllisGit, 2020-08-16:
Hi @spanzetta
which version of OP and Python do you use? Maybe the issue is related to newest OP-Release.

In my dev-environment I can't reproduce the error.

OllisGit, 2020-08-17:
Hi, 
I looked into the sourcecode and currently I don't have an explanation how this could happen.

In the newest version 1.23.0 I added more log - informations in the plugins event-log.
Please activate the event-logging in the plugin settings and if the error occurs, please download the ```plugin_DisplayLayerProgress_events.log``` file and attach it to this issue.
[image]

Thx, in advance
Olli

allenrowand, 2020-08-18:
[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5087382/plugin_DisplayLayerProgress_events.log)
[image]

Thank you for following up on this, let me know if there's anything else I can do to help.

OllisGit, 2020-08-18:
hmm... thanks for the log. I took a rough look into the file and I think I found the issue... but at the moment I have no idea how to solve it.

In the last versions/month I captured a lot of new events with a lot of "cool" values which could be send to the display.
Because of increasing the amount of events the performance of the OP-Server decreased (could be the reason for layer-shifts).
So, I changed the processing behaviour to a queuing async-way. Now the performance is good, but each display event is processed after each other and creates a delay between occurring and displaying.

In your case:
```
-- Received the 1% progress changed event
16:29:38,939 ON_PRINT_PROGRESS: 1 [1704100960::Thread-61]
...a lot of other events
-- Send the progress change to the display
16:34:44,510 SEND-COMMAND: M117 1% L=-/100 40.0mm [1704100960::Thread-63]

--> delay of 5 seconds
```

Can you confirm a delay of ~5 seconds?
Please also provide the octoprint.log.

Btw. turn off the event log, because it could create huge log files.

allenrowand, 2020-08-18:
It's only a second or two between I see the progress change in the state sidebar and when it updates on the printer's display. The issue for me is the plugin shows a completely different progress value than the progress bar; in the screenshot above the plugin is reporting 1% completion while the progress bar shows 17%.

allenrowand, 2020-08-18:
Here's my octoprint log; there's now a ton of error messages regarding notifications. I don't recall seeing this before I upgraded DLP to version 1.23.0
[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5092294/octoprint.log)

OllisGit, 2020-08-20:
Hi @allenrowand ,

you use a lot of other plugins, did you try to deactivate other plugin to see an improvement/change. E.g. disabling ActionCommands, produce a lot of noise....and btw. I looked into the implementation and the author did the same mistake I did in the past. The plugin use a OP-Hook in the communication layer.
https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-action

>Warning
>
>Make sure to not perform any computationally expensive or otherwise long running actions within your handlers as you will effectively block the receive loop, causing the communication with the printer to stall.
>
>This includes I/O of any kind. 

But maybe this is not related to the current issue.

allenrowand, 2020-08-20:
I removed a few plugins, but if I unhide the progress bar it still shows a different completion percentage from what DisplayLayerProgress shows on the printer display, browser tab, and state area of OctoPrint:
[image]

My expectation is that the 90% shown by DLP would be the same as the 96% in the progress bar. Is your code generating the blue progress bar, or is that value coming from someplace else?

I'll file a bug re: ActionCommands spurious logging.

bvrielink, 2020-09-03:
Got the same problem. At the moment a 21% difference (73% instead of 94%). Got a lot of plugins but most are for visual stuff etc. Only plugin which could be a problem is maybe PrintTimeGenius. Unfortunately I am not willing to disable plugins but I can help with debug or logging of needed.

OllisGit, 2020-09-03:
Hi @bvrielink ,

next time you print please do the following:

- turn on event-logging in DLP-Settings
- check size of octoprint.log sometimes it is big and should be delete before a debug-print starts
- restart server
- start print
- after print. please attache two log-files to this issue (octoprint.log and plugin_DisplayLayerProgress_events.log)
- turn off DLP event-logging

Thx,
Olli

bvrielink, 2020-09-04:
> Hi @bvrielink ,
>
> next time you print please do the following:
>
>     * turn on event-logging in DLP-Settings
>
>     * check size of octoprint.log sometimes it is big and should be delete before a [...]

Hi @OllisGit , 

See attachment for logging. I did a slight search/replace in the OctoPrint log file btw (ip's and usernames)

It's from a cancelled print which ran for a couple of hours. At some point the difference was huge. It showed for example 12% but it was already at 62%. All other plugins like M73 Progress (which I only installed today) transfer the correct percentage to the printer. Makes me curious where the percentage for this plugin comes from. 

Hope you can find the problem. Let me know if you need more info.

[layer.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5175429/layer.zip)

spanzetta, 2020-09-05:
problem still present,,, I hope it can be fixed by the author..

spanzetta, 2020-09-10:
Is there any news on this topic?
Thanks a lot

bvrielink, 2020-09-10:
When I look at the code this plugin does not factor in plugins which change/improve the progress (as PrintTimeGenius does for example). If I look at plugin Progress Title (https://github.com/pwnbus/OctoPrint-Progress-Title) which does show the percentage correctly it does take this into account:

                   if (self.custom_time_plugin){
                        // References the PrintTimeGenius plugin
                        percentage_completed = (data.printTime||0) / ((data.printTime||0) + (data.printTimeLeft)) * 100;
                    }

Hope @OllisGit can change this. If not I will make a fork as I quite like PrintTimeGenius.

spanzetta, 2020-09-10:
> Hi @spanzetta
> which version of OP and Python do you use? Maybe the issue is related to newest OP-Release.
> 
> In my dev-environment I can't reproduce the error.

Hi @OllisGit 
Here is my details..

Raspberry PI 3A+
OctoPrint 1.4.2
Python 2.7.16
OctoPi 0.17.0

Waiting for your feedback... and I hope you can fix the problem.. it happens always/consistently..

OllisGit, 2020-09-19:
Question: Who is using "PrintTimeGenius"? If you disable this plugin, is the lack gone?

OllisGit, 2020-09-19:
Hi @bvrielink,
I looked into the ProgressTitel and PTG implementation and both overwrite the viewmodel behaviour of the progressbar  to include the PTG values instead of the system progress value.

Now the question is, how could this UI values send to the DLP-Plugin, so that the plugin could spread this info to: the printer-display and the other UI fragments that DLP supports (also the EventBus and the public API)?

Do you have any ideas how your PR could look like? Ideas welcome!!

allenrowand, 2020-09-19:
> Question: Who is using "PrintTimeGenius"? If you disable this plugin, is the lack gone?

I use PTG, will test as soon as I can.

bvrielink, 2020-09-19:
> PrintTimeGenius

I am using PrintTimeGenius as well. Before I used PrintTimeGenius there was no lag. But the print estimates were not correct because the estimates from OcoPrint are very inaccurate that's why I use PrintTimeGenius which solves this problem.

bvrielink, 2020-09-19:
> 
> 
> Hi @bvrielink,
> I looked into the ProgressTitel and PTG implementation and both overwrite the viewmodel behaviour of the progressbar to include the PTG values instead of the system progress value.
> 
> Now the question is, how could this UI values send to the DLP-Plugin, so that the plugin could spread this info to: the printer-display and the other UI fragments that DLP supports (also the EventBus and the public API)?
> 
> Do you have any ideas how your PR could look like? Ideas welcome!!
> 
> Currently I am using this "backend" function, but this could only deliver the system-progress and not the PTG progress:
> 
> ```
> def on_print_progress(self, storage, path, progress):
> ...
> ```

Hi @OllisGit ,
My suggestion would be to look if a linear time plugin is installed and if so don't use the system progress but calculate it like in the example above. Took a look a couple of days ago and I believe only printTimeLeft was missing atm in your plugin. If that is available it's not a lot of work for the display part but to be honest I didn't look at the API and EventBus stuff so maybe that's another story.

OllisGit, 2020-09-19:
fyi: I found a way to solve the issue, but I need to do more testing.

bvrielink, 2020-09-19:
> 
> 
> fyi: I found a way to solve the issue, but I need to do more testing.

Great! If you want me to test as well let me know.

spanzetta, 2020-09-20:
@OllisGit 
That's a great news!
Let's wait the results of your testing..
Thanks

OllisGit, 2020-09-20:
Hi,

new version 1.23.3 is out.

If PrintTimeGenius is used, the progress is calculated instead of using the system progress value.
Now there is no huge gap between progress bar and display. Sometimes the progress is is slightly ahead, maybe a rounding issue or a higher update interval. But I think this is acceptable.

Please test and give feedback.

BR
Olli

bvrielink, 2020-09-20:
Hi @OllisGit ,

Did a quick test and during the print it seems to be okay now. Sometimes a percentage above like you said but that's no problem. What is a pity that it only updates the percentage during the actual print. My print progress was already at 10% because of heating and abl but the progress stayed at 0% until it actually started printing (then it jumped from 0% to 10% immediately). Don't know if you want/can fix that but that's at least my observation.

spanzetta, 2020-09-20:
I confirm that now it works almost perfectly..
Good job!!!
Thanks

closed 2020-09-20 by spanzetta

allenrowand, 2020-09-20:
Confirmed to be much better here, thank you! Now I'm removing the text display from the progress bar; I have Themeify set .progress-text-front to display:none
[image]

spanzetta, 2020-09-23:
Sorry to come back here.. but I have a question even if not directly related to this plugin..
Now all percentage shown is correct.. but they are not in the notifications sent by both PushBullet and Telegram plugins (eg. it is printing at 30% but the notifications says 7% )
Do you have an idea on where the wrong data is read by PushBullet/Telegram plugins?
Should we report this issue to their author?

Thanks

OllisGit, 2020-10-02:
Hi @spanzetta ,

the problem is, that the PTG-Plugin overwrite the time-calculation stuff in OctoPrint, but not the progress-calculation. Thats the reason that each plugin needs to adjust the progress by it's own, like I did or the OctoPrint-Progress-Title did.

So, you need to raise an issue for each plugin that use the progress-value.

spanzetta, 2020-10-02:
Ok 
thanks for the clear explanation